vlachoudis/bCNC #6: Bad path?

pommen, 2015-02-13:
Hi!

This project looks really cool and I want to try it out.
Been looking at this this a bit trying to get it to work. Now, please bear with me, I'm not a Python user. 

I have installed the prerequsits, but when I try to start bCNC.py i get a command window telling me that rexx is missing(Had to record the error and play it back slowly to see the error message, if there's another way please share, im a total newb). 

Copying all the lib to root lets me start the program, Wich looks awesome, btw =) But i couldent connect to the arduino.

/Peter

pommen, 2015-02-14:
Opps, asked and answered before, sorry.

closed 2015-02-14 by pommen

vlachoudis, 2015-02-15:
You have the start the program using the script that sets the path for the libraries.
If you are running on another system you can set the PYTHONPATH variable pointing
the directory of lib to find the additional libraries.
Did you manage to connect to arduino?

pommen, 2015-02-15:
Yes, i worked perfectly when I path'ed it. There's tons of articles on how to do it, but to recap what i did:

My Computer > Properties > Advanced System Settings > Environment Variables >

Then under system variables I create a new Variable called PythonPath. In this variable I have C:\Python27\Lib;C:\Python27\DLLs;C:\Python27\Lib\lib-tk;C:\CNC\bCNC-master

Beeen using to mill out a couple of PCB this weekend with great results :+1: