vlachoudis/bCNC #7: Problems Sending Gcode after probe

crowjik, 2015-03-17:
Thanks for putting out this program!

I have had two issues which are seemingly unrelated but maybe not.

First was while setting up the program. Twice I ran into an error that was something like:
"bad event type or keysym "KP_delete"
Once this pointed me to CNCList.py line 35, and once it pointed me to bCNC.py line 884. Both times I was able to get past the problem by commenting out the relevant line. This happened to me on both a Windows 7 and a Windows 8 machine, with both Python 2.7 and 3.4.

Once I commented out those lines the program seemed to work fine. I have been able to load Gcode, mirror the geometry, send manual commands, and run the probe scan. However, after running the scan and zeroing one of the points of the matrix I was unable to run the Gcode program. 
In the bCNC window, on the status bar, I get an "error: not idle" message
In the bCNC terminal, it looks like it has sent G90 from the Gcode, gotten an ok response, then stopped
In the command prompt, I get the following message:
File "c:\Python27\Lib\lib-tk\Tkinter.py", line 1532, in __call__ return self.func(*args)
File "c:\Program Files (x86)\bCNC\bCNC.py", line 2959, in run lines,paths = self.gcode.prepare2Run()
File "c:\Program Files (x86)\bCNC\CNC.py", line 1725, in prepare2Run if self.gcode in (1,2,3):
AttributeError: GCode instance has no attribute 'gcode'

Looks like a great program if I can just get past this. Any help or pointers are greatly appreciated.

vlachoudis, 2015-03-17:
Thanks for your report.
I have recently modified the program to include a basic CAM function for profiling, cuts and box-with fingers creations (for the moment available from the command line), and apparently I introduced a bug in the probing.
I have corrected the errors you send me, and should work now.
Could you please test it and let me know.

vlachoudis, 2015-03-17:
Apparently the numpad TK_ events do not exist on the windows python tkinter version.

crowjik, 2015-03-17:
The numpad thing doesn't really affect me anyway, so no worries on that. I'm downloading the new version now and will try it today - I'll let you know how it goes.

Thanks!

crowjik, 2015-03-17:
Working great now, thanks! I noticed two more minor bugs but I'll open up another issue for them.

closed 2015-03-17 by crowjik