vlachoudis/bCNC #9: Single Probe Error

crowjik, 2015-03-18:
Last one :) When I click on the "Probe" button to run a probe at a single point, the program returns a GCode error "Invalid GCode". I see in the terminal that the code that is sent is "G38.2 F100" (assuming I've set the feed at 100). I can manually send the command "G38.2 Z-1 F100" and it works fine so I guess you just need to add a maximum Z depth for the probe command into the program, so the button sends that.

vlachoudis, 2015-03-18:
You should provide a value in the Z field of the probe otherwise it doesn't append any value in the command. Maybe I should somehow protect it.

crowjik, 2015-03-18:
I see now. Maybe not protecting it, maybe just rewording the message that comes up when you hover over the x, y, and z fields. Something like "X Coordinate for single probe", "Y coordinate for single probe", and "Maximum Z depth for single probe operation" ? I hadn't understood that those fields were the probe coordinates, even though maybe it should have been obvious because of their location in the gui...

vlachoudis, 2015-04-05:
I've added an error messagebox popping when no direction is chosen

closed 2015-04-05 by vlachoudis