vlachoudis/bCNC #13: Pendant requests.

chamnit, 2015-04-22:
@vlachoudis : New issues thread on the web server pendant. I have to say I keep uncovering amazing features that you have every time I use bCNC. This pendant approach is an amazing solution. It has made my day. This coupled with an RPi installation could make Grbl and personal CNCing so much more accessible.

A couple of comments/requests on this: 
- The homing, unlock, and reset buttons are bit close to other buttons that would be used during a job. I haven't checked what happens to the pendant when I job is run, but I think this could easily lead to accidental button pushes. The solution could be as simple as moving that row of buttons above the status and position table.

- Could you add an MDI command line somewhere such that custom individual commands can be sent from the pendant? For example, after homing, I send a G28 command to quickly move the spindle to the center of the work volume.

chamnit, 2015-04-23:
@vlachoudis : I also tested the pendant as well. There were some bugs I did notice:

-  It was quite laggy as well. A bit slower than the main GUI  buttons. They didn't operate very well and got stuck in different states for some reason. It can't tell you if that is the result of the RPi's slow CPU or not yet. I will be testing against my Macbook Pro and try to reduce some of the CPU load on the RPi by disabling all of the visualizer options.

- The jogging distances were exactly (3x) the increment selected in the drop down button menu. It doesn't seem related to inches or mm mode due to the multiplier exactness.

vlachoudis, 2015-04-23:
@chamnit I've implemented your suggestions, moving the Home/Unlock/Reset buttons above the status and add a command line. Even though the G28 was hidden inside the WGS settings :)
I never used this command before, but thanks to you I will try it out!

Connecting from my Nexus7 on my linux box (a very old (10y) laptop that I use for my CNC) the buttons are operating normally without any lag. I don't know if it is related to the RPi or the telephone you connect from. Maybe its a question of multiple threads and synchronization. I open one new thread for the server and one for every connection (unfortunately). Once it receives a message it appends it to a Queue() and will be handled from the serial receiver timed method (every 250ms) in the main program.

The jogging for me behaves correctly. It cannot be inches or mm, the program is agnostic on the units. The index.html page prepares a gcode and it sends it to the device. Could it be that your telephone browser is sending multiple commands. If you want I could put a dump on the screen of the received commands so you could test it, since I don't have any RPi to test.
Or you could do your self if you prefer. In the CNCPendant.py in the do_GET method there
is a "elif key=="cmd"", just add "print value" below.

chamnit, 2015-04-23:
@vlachoudis : Thanks! I missed that G28 was in the WGS settings. It's typically used to move the machine to a ready position after homing, like I said. The other one, G30, is used for moving to a set tool touch plate location to determine tool length offsets and to automate tool changes. (We could discuss that another time on how to implement that. I haven't yet written up the recommended way to do it.)

I'll check if the lag is coming from the Wifi dongle on the RPi and the overhead involved with using VNC. I'll connect everything to ethernet and a monitor tonight.

I also forgot to check if the RPi was running in hard float. (I need to start making a list! :) )

chamnit, 2015-04-24:
@vlachoudis : Pendant testing results.

- My iPhone 5s with its Safari web browser was indeed sending 3 commands per button press. The "print value" addition clearly showed this. I tested this with Chrome on the iPhone. Chrome only issues 1 command per button press, but was incredibly laggy. It was on the order of seconds for each button press and I have no idea why. According to the "print value", the lag was between the button press and when it showed the value print out. I did not test on a desktop browser to see if the behavior was limited to the smartphone. I will later this weekend.

- The MDI line in the pendant worked great. Thanks for that addition. The layout was a little messed up in portrait-mode on the iPhone, while landscape looked right. What do you use for your pendant? A tablet?

I'll continue testing over the weekend. Let me know if you want to look at anything in particular.

vlachoudis, 2015-04-24:
@chamnit Why safari is sending 3 commands??

I was using my Nexus 7 tablet in portrait mode. I don't use any longer the pendant since I connected an old laptop next to my CNC.

bCNC is opening a web server, serving all .html pages in the directory of bCNC. One option could be to write a second page one for portrait and one for landscape.
Another option would be if the javascript is able to detect the orientation to change the page
However I am newbie in javascript, so I would need some time to understand how to do it.

chamnit, 2015-04-24:
@vlachoudis : That's a good question that I have no answer to. I recall seeing some issues related to the touch interface and perhaps a bug in iOS and mobile Safari. I'd probably think it's due to the latter.

After some reading, the touch interface on smartphones and tablets have this standard lag to separate a single tap from a double tap for zooming. On iOS, it's about 300ms. I think this is the primary lag that I'm seeing, especially when jogging. It's impossible to quickly button press and the lag is slow enough to accidentally press more than once or skip one by the interface zooming in. 

Perhaps, the virtual pendant isn't quite the answer.. Or at least, the implementation isn't conducive to how touch interfaces work. I'll have to mull on that over the weekend. It's almost as if you need a native app to get around these problems, unless there is a way to force touch-based web-browser to become more responsive and ensure cross-platform compatibility.

vlachoudis, 2015-04-25:
@chamnit I've tried it again the pendant with the nexus 7 and the galaxy s3, both are quite responsive, however what I don't like is that you don't have any tactile feedback before pressing the button to see if it is the correct one.

One option would be to check if it is the "vibrate" function when a key is pressed or the setInterval() for getting the status in the index.html, is causing the multiple clicks on Safari

chamnit, 2015-05-09:
Remembered to check if the vibrate() command in the index.html was the cause of the three button push problem with the Pendant and an iPhone. No change. It still commanded everything 3 times...

vlachoudis, 2015-05-09:
I found on the web https://github.com/ftlabs/fastclick
To eliminate the 300ms delay, maybe could work for safari?

chamnit, 2015-05-10:
@vlachoudis :  I don't have much web programming background, but I'll look into it.

chamnit, 2015-05-10:
@vlachoudis : Tried to get something to work, but failed in most respects to get any different behavior. I did verify that Safari on my Mac does the 3 clicks per button push, while Chrome on my Mac doesn't. It seems it's something with Safari itself.

jarretluft, 2015-05-13:
@vlachoudis I took a look at trying to port the index.html file to use jquery and a basic css framework to simplify the ajax requests and take advantage of all the different browser specific issues that jquery corrects automatically.
I discovered that I cannot reference the jquery.js file from inside the index.html file, presumably due to how you have set up the basic http server.  Is there a way to modify this server so that something like this <script type="text/javascript" src="jquery-2.1.4.min.js" /> would work, and not fail to load the jquery framework?

I noticed above that you said the basic http server would serve up any .html web page, is there a way you can expand that to also include .js and .css file types, so that a more advanced pendant/browser based gui could be developed?

let me know what you think. thanks!

vlachoudis, 2015-05-13:
@jarretluft I am not aware of the jquery.js. If it is one javascript file, you could place it in the same folder of bCNC. The basic server of the program should serve it as
http://localhost:8080/jquery.js

FYI (maybe I should write on the wiki)
Normally the server I have accepts a couple of special requests:
/send?cmd=xxx     as something being typed from the command line
/send?gcode=xxx   send gcode directly to the machine, multiple lines with %0D
/state                    return the position and state of the machine as json
/config                   <incomplete>, for the moment returns the rpmmax of the spindle
/icon?name           serve any icon name.gif file from the icons folder
/                           return the index.html
<file>                    return any file in the bCNC folder.

jarretluft, 2015-05-13:
Thanks for the info. In my testing I tried accessing it at http://localhost:8080/jquery.js and it was returning the error page you have defined [here](https://github.com/vlachoudis/bCNC/blob/master/CNCPendant.py#L111)

I will take another look though, perhaps I was entering the file name incorrectly in the URL path...

vlachoudis, 2015-05-13:
Indeed there was an additional "/" infront of the name resulting in a wrong filepath.
Now it is fixed if you download the last version the repository

jarretluft, 2015-05-13:
Thanks Vasilis, I will give the latest version a try tonight.

chamnit, 2015-05-29:
@jarretluft @vlachoudis : Where did we get with the virtual pendant and the web socket implementation? Were you two waiting on me for something? I can't recall.

jarretluft, 2015-05-29:
Nothing from you, I have some more work I wanted to do on it before I send
@vlachoudis a pull request, but have had limited free time in the past
couple weeks.

vlachoudis, 2015-05-30:
This was my understanding as well. I am waiting for @jarretluft.

Effer, 2015-05-30:
In view of the pull request may I suggest to move all pendant stuff (js,html.css etc) to a separate folder? I believe the web part will grow more in functionality and will be easier keep it separated from python scripts.

vlachoudis, 2015-05-31:
@Effer A  separate directory is a good suggestion. Also to move out the javascript code from the index.html in a separate file.

Effer, 2015-06-15:
@vlachoudis I'm working on a [commit ](https://github.com/Effer/bCNC/commit/9ac0fe798db836e4b38979db90814170905b358f) in my fork for uploading the file directly from the pendant web interface. 
Would you like to have a look if it could worth a pull request?
It's still a draft but I suppose it would be usefull to have a way to move file to bCNC streamer from CAM workstation without using any media or shared folder.
Let me know.

chamnit, 2015-06-15:
@Effer : I think @jarretluft was working on uploading a file directly from the web pendant itself as well. He was porting [GrblWeb](http://www.github.com/andrewhodel/GrblWeb/) to work with the Python backend. He needed this to get JSCut to work as an add-on. Not sure how far he got with this.

Effer, 2015-06-15:
Oh well, :(
Let's say It helped me to learn something new...

vlachoudis, 2015-06-15:
@Effer I was unaware the @jarretluft was working on that. Maybe we need to synchronize a bit the development.

Effer, 2015-06-16:
@vlachoudis me too :)
I think as soon the macro step will be included many issues will close, and that will help to decide what would be more useful to develop.
Anyway also in my little addiction to improve the pendant interface I met some decisions that would have required a comparison.
I'm discovering contribute to a open source project is quite complex.

vlachoudis, 2015-10-14:
@jarretluft Its being some time I didn't use the pendant, but I did last night and I've notice that after the restyling the state doesn't change the color any more. How can we fix it? Also from the wcs input fields the type="number" was removed. I restore it back since this one opens the numerical keypad on the phone.

Nate-Bro, 2015-11-08:
Last week I finally finished my first cnc machine. I tried a few gcode senders, but came across this project today. I have not had a lot of time to play with it, but from what I have seen I believe I have found my gcode sender.

I would also like to contribute. I have not worked much with python/graphics, but I have been a web developer for awhile now and think I might be able to help out with this Web part.

Effer, 2015-11-08:
Hi @Nate-Bro !
You made the right choice :)

Any contribute is always well accepted. @vlachoudis have just [commit](https://github.com/vlachoudis/bCNC/commit/ca5ac1325bd01c4abd5ff90d03c042bd8c8d1392) to improve the web pendant. So you arrive just in time to improve his solution. We lack of web developers.

jdiderik, 2015-11-08:
@Nate-Bro Same here, not very experienced in Python, but webdeveloper for a long time.
Maybe we can team up and take the pendant to the next level.
I've looked into Websockets in Python to hook up to the bCNC interface features like the DRO and jogging, to give the pendant more 'live' updates and reaction time. The 'slughish' feeling currently what prevents me from using it.

@vlachoudis @Effer (and others)
I love bCNC, not only straight out of the box, but the flexiblity with the configuration through the config file too (since IO don't speal Python that well, Added the 'Run' ribbon button to all other menu items and moved the 'Terminal' menu to the left, created larger icons. I have a current set up with 1920x1080 so it's easier to use with larger buttons and always have the ability to pause-resume-sstop what I'm doing. (Although I have just finished wiring up my hardware Pause-Resume buttons, making live even easier)

chamnit, 2015-11-08:
@Nate-Bro & @ jdiderik : I would love to see the web interface be pushed more. There's so much you can do with it, as a pendant or as a stand-alone interface. I think the most important thing here is to make sure that we keep things modular like Vasilis has designed bCNC to be. This creates more powerful and flexible tools down the line.

I do know that @jarretluft was working on adding web sockets to bCNC (See his [repo](https://github.com/jarretluft/bCNC)). One of Vasilis' requirement is not having to install any additional Python package. Jarret found a websocket library that exists as a file. It should be in the lib folder. I can't say for sure how well it performs but it's a great start.

Nate-Bro, 2015-11-08:
Sounds good, just let me know what the plan is, and what exactly needs help.

vlachoudis, 2015-11-08:
@Nate-Bro @jdiderik you are both more than welcome to contribute. The web pendant needs quite some work :) and fresh ideas. @chamnit is correct I would like to keep the dependencies down to absolutely minimum.
A few things that needs re-working:
- introduction of web sockets to have a continuous tcp connection
- camera viewing. I've tried to do something something like what @jdiderik #136 proposed with static images and opencv (still very basic)
- beautifying the interface would be nice. e.g. the state-color is no longer working :), alighment of controls etc...
 enabling and disabling the buttons depending on the state of the machine/running
- I've tried to split the interface from the sender functionality (is not fully implemented) so one could start only the web pendant without the GUI.
- and many other...

Nate-Bro, 2015-11-09:
I was using it yesterday and was thinking how handy a camera viewing feature would be

chamnit, 2015-11-09:
@Nate-Bro : I'd recommend looking at Andrew Hodel's GrblWeb's [source code](https://github.com/andrewhodel/grblweb). He generally writes clean JS code and has a web-cam feature installed there. GrblWeb is very nice too, but doesn't have nearly as many features as bCNC does. I think if we can mash the two together (and add the features bCNC has into it), this would be the beginnings of a great web platform for bCNC.

jarretluft, 2015-11-09:
@chamnit , @Nate-Bro : I had been playing with integrating grblweb with bCNC via websocket a number of months back and had made some progress, at least with all the basic CNC control.  I still have the code on my machine locally but never pushed it to my git repo.  I don't have the time to continue working on it at the moment but I will try to push my work to date to my repo and you can look at moving forward with it if you would like.

Nate-Bro, 2015-11-10:
@jarretluft please do add to your repo. I look forward to seeing it.

jarretluft, 2015-11-11:
@Nate-Bro ok, the repo is updated at: https://github.com/jarretluft/bCNC
Note that @vlachoudis has added 188 commits to his branch since my work was done on this, so there is likely a bunch of new code/features in bCNC since my work that could change how the web pendant interacts with the back-end in newer versions.

Also, I noticed the websocket_server.py script that is the basis for the websocket in the lib folder works well when using python 2.7, but was crashing for me on OSX when I tried in python3 (something about having to encode strings as binary in the socket library).  So just beware that if you have trouble running my repo on python3 I wasn't able to track down what was causing the error but it should probably be fixed at some point :)

Nate-Bro, 2015-11-16:
Sorry it's been a few days,  I moved my cnc machine out to the shop and have been working on configuration of software.

Turns out I uploaded the wrong he file to the board and was getting some interesting issues.

However I still have issues with the auto leveling. But I'm sure that's all me and not a software/hardware issue.

I'm going to start playing with the pendent  this week.

Just a couple things, first has anyone done a progress canvas for the pendent like what is in the main window?  I have not played with anything yet, but seems like the gcode could be converted to an svg, and completed lines have a color change like the main window.

Also what is the best way to  keep in contact, these comments or email or.....    thanks