vlachoudis/bCNC #14: Pendant Server Address/Port

Timvrakas, 2015-04-23:
In my opinion after the pendant server is started, the Address and Port should be permanately displayed somewhere in the GUI. Even a line in the terminal window would cut it. (no pun intended!)

Also (and more importantly) the address for the pendant server should never be "localhost" (unless there is no internet. For example I need to know what IP the school WiFi has assigned me, so I can connect. Localhost does not help much.

chamnit, 2015-04-23:
@Timvrakas : The address is easy to find with some terminal commands like ifconfig on Macs and Linux. And I believe the port is always set at 8080. I would also check if your school blocks non-standard ports.

Timvrakas, 2015-04-23:
Oh yes, I know how to find the IP address. I just though that the window
that pops up with the 'started on:' message should contain the real IP
address, not just "localhost". It just save one step. I do have the pendant
working, but the IP address pool is changed every day, so I have to go
check each time.

Thanks

vlachoudis, 2015-04-23:
@Timvrakas your suggestion is a valid one, unfortunately even if it seems strange, python doesn't have a method in the socket class to return the IP address of the local host but the hostname. The problem is that each machine can have multiple network interfaces. My machine typically has 3, the loopback, wlan0 connected to my router and the tun0 for my VPN connection. Moreover the address that my router is assigning is an internal address in the space 192.168.1.x which is not visible from outside the router.

I was looking on the web [http://stackoverflow.com/questions/166506/finding-local-ip-addresses-using-pythons-stdlib], there is a python class to return the interface, but it is not a standard package in python, maybe it depends strongly on the OS.

I could have even called the "ifconfig" and report all the possible IP addresses, but ifconfig is on linux based machines. On windows I think is the "ipconfig" with a different output.

I found once a method that was based on connecting on a remote web site that the only function it does is to return the IP address, but this requires external connection.

I will try to investigate it a bit more.

closed 2015-04-28 by vlachoudis