vlachoudis/bCNC #16: zero out coordinates?

lordmundi, 2015-04-24:
One other question.  In most gcode senders there is a way to zero out an axis.  For example, I routinely have to find a corner of my workpiece and make that my 0,0,0 point.  I usually do this by going to the corner and hitting a "zero" button which sets the current coordinates to zero (I think using G92).  I've looked through the GUI and don't see anything like this.  Do you have to manually go to the WCS page and type in what you want and hit set?

chamnit, 2015-04-24:
Yes. You'll need to fill it with zeros and click set as far as I understand. It applies to the G54-59 work coordinate frames instead of using G92.  It's not how you're probably used to it. Other Grbl GUIs usually use a G92 button to zero, which isn't recommended because it isn't persistent.

But I would agree a click button to zero one axis or all axes that get applied to the G54-59 work coordinates would be a good thing to have. It's typically implemented with a small '0' button next to each axis value of a WCS, followed by a large '0' or 'zero all' button that zeros all axes of that WCS.

lordmundi, 2015-04-24:
i agree, and I would put it on the front jog page!  just some way to zero each axis independantly.

great stuff so far. Nice and light.  I have a jog dial project that right now works with chilipeppr, but I think you could easily make my hardware jog dial work with this also.  If you're interested, you can check it out here: https://www.youtube.com/watch?v=2wUOFlnGSMA

vlachoudis, 2015-04-24:
Its a good point. Actually I had it in my first implementation and then I removed it since I was accidentally hitting the wrong axis button to zero with the mouse. So I though an extra step of complexity will help me to avoid zeroing the wrong axis :)
I've put it back, 3 extra buttons X=0, Y=0 (for the moment in the WCS tab). You can check the latest svn version.

Just for information. You can set the zero also from the command line by typing
either:
set0
set 0 0 0
setx 0   (equivalent to set 0)
sety 0 (equivalent to set * 0, where * is any non number character )
setz 0  (equivalent to set , , 0)

the command line is easily accessible from everywhere by hitting "Space" anywhere or "Ctrl-Space"

closed 2015-04-28 by vlachoudis

vlachoudis, 2015-05-07:
@lordmundi I completely forgot you can zero the coordinates from the main page by just clicking on the coordinate you want to zero (it behaves like a button)

chamnit, 2015-05-07:
@vlachoudis : Hmm, that could be potentially dangerous. An errant click could screw up a users job if they did not know it was there. Is there a way to make it more obvious?

vlachoudis, 2015-05-07:
@chamnit I am afraid you are right. I will remove it, anyway I don't believe anybody was using it, neither me :)