vlachoudis/bCNC #17: Failing to run -- divide by zero error

1bigpig, 2015-04-24:
I am sure I am doing something wrong but when I try to start bCNC, I am getting this error:

pi@raspberrypi ~/Downloads/bCNC-master $ ./bCNC
Traceback (most recent call last):
  File "./bCNC.py", line 3540, in <module>
    application = Application(tk)
  File "./bCNC.py", line 794, in __init__
    self.probeChange()
  File "./bCNC.py", line 3080, in probeChange
    self.probeXstep["text"] = "%.5g"%(probe.xstep())
  File "/home/pi/Downloads/bCNC-master/CNC.py", line 143, in xstep
    self._xstep = (self.xmax-self.xmin)/float(self.xn-1)
ZeroDivisionError: float division by zero
[1]+  Killed                  ./bCNC
pi@raspberrypi ~/Downloads/bCNC-master $ 

Any ideas?  It use to run and the only thing that I can think of, is I updated opengl to try to get Universal G-Code Sender to show the visualizer...

Bruce

vlachoudis, 2015-04-24:
You have used 1 step in the x-probing grid with a result of a division by zero.
I've just updated the code to protect for this situation
Can you download the new svn and test

1bigpig, 2015-04-24:
That fixed it.  Thanks!!!

I have some other questions about program, but I need to organize them and I will post them up in a separate section.

Thanks for fixing it so quickly!
Bruce

closed 2015-04-24 by vlachoudis