vlachoudis/bCNC #19: Metric/Imperial bugs

1bigpig, 2015-04-29:
After using the program for a few days, I have found a couple of issues that are metric "correct" but not if you are using imperial measurements.

First off, the tool path follower lights up correctly, but the tool animation does not (it only moves the metric amount) instead of the imperial amount.

Second, the probe feature, while it does convert the imperial measurements to metric, will not nativly accept imperial measurements.  

The probe grid and the material grid are only in metric and while I can get the probe grid to resize, I have not been able to change the settings for the material grid.

Also, the WorkArea is always negative and about 10 times larger than the Material grid.  Is this by design and/or is there a way of changing the WorkArea to represent the Machine Tool work area?

I don't know if this is GRBL or bCNC, but it ignores an M30 as a program end.  GRBL says it is supported but Universal G-Code sender and bCNC are ignoring this command as a program end.

Lastly, I guess this is confusion on my part, but what is the difference between the program "Pause" and "Stop"?  "Pause" seems to stop the program instantly but will resume when "Pause" is pressed again.  "Stop" seems to no stop but just quit sending G-Code.  Which would be ok, except that usually the buffer still contains about 15 moves.  If you press "Pause" to stop the machine in case of a minor emergency (work piece came out of the vice, tool breaks, tool about to hit the vice, ect.), if you then press "Stop" the stop command is ignored.  If you just press "Stop" the machine seems to continue on its merry way until the buffered commands are executed...  So far, the only solution is to press the soft reset, but this is not ideal as it loses the G92 offsets...

Am I doing this wrong, or is there another way to implement an emergency stop that stops the machine, clears the buffer, and preserves the G92 offsets?

Thanks and even though I am finding bugs, I am REALLY digging your program.  I am running it wirelessly on a RaspberryPi from a browser window and it allows me to run my machine without tying up my design computer.

Please keep up the good work!!!

Bruce

PS I am thinking about making a script that takes a virgin RaspberryPi install and adds all the parts to make it work headless with bCNC.  Linux is not my specialty but I figured I know enough to make a simple script for people who do not know Linux but want a cheap headless CNC controller that is accessible from any browser window.

vlachoudis, 2015-04-30:
Thanks for the report. Its true I have never test it with imperial as here in Europe we (apart some exceptions) only use metric.
I am sure you have both the correct settings in the init file ~/.bCNC and the same in grbl setup.
I am bit puzzled since the program it self should be independent of metric/imperial apart from the recent addition on the "Tools" tab, where possibly there to be issues between metric/imperial.

1. The display, probe, jogging, gantry location they know nothing about the units they simply show the numerical position (maybe I didn't understand your point)

2 The default dimensions of the workarea are negative (WPos). I made this assumption since this was what I was getting from grbl after homing. Maybe I am wrong there, I need to understand a bit better. The dimensions of the work area you can change in the .bCNC ini file or from the "Tools" tab "CNC" section, but then you have to restart the program

3. There is no correlation between work area and material grid. The grid is only displayed for the dimensions of the work piece

4. M30  I don't know what this command is doing :). bCNC sends all the commands to grbl as such

5. Stop,indeed there is bug there. I have to figure out the correct way of stopping the program and clearing the buffer. Now it sends a feed hold, but as you correctly mention the buffer is not empty.
I have to read the grbl documentation on how to do it. Maybe a soft reset could be the solution

6. The G92 is not persistent. You should zero the machine using the workspaces. Go to the WCS tab and select one of the first workspaces and zero the coordinates. This is persistent across hard and soft resets.

If you like we could include your RPi scripts in the github

1bigpig, 2015-05-01:
Ok, after using the program all day, I might have tracked down the metric/imperial discrepancy.  I think it might be a drawing canvas setting.  The drawing has obviously been multiplied by 25.4, but the tool animation is not.  Again, according to some of the usage reports I have heard/seen from Sonny Jeon, metric usage is like 99%, so you might be fine just ignoring this particular issue.

M30 is sorta the same as M2 (program end) but I think M30 includes a "rewind".  The only reason it is even an issue, is my CAM program uses it as a program end.  The behavior I am having is that bCNC keeps trying to send commands--or at least thinks that the G-code program is running.

While I understand that G92 is not persistent, it is what I learned to use (for right or wrong).  I just need to figure out an easy way to set the Z axis as I use a tool touch off place for setting the Z axis.  My touch off plate is 3.07mm thick, so I cannot just zero it out.  

I did figure out how to get the work area to be positive.  I went into the .bCNC.ini file and lied to the program and told it my machine was negative size and that seems to put the WorkArea in the correct coordinates...

After using the RaspberryPi all day running a few pieces, I am beginning to think that a stock RaspberryPi is not really faster enough to even act as a front end controller.  At the stock speed, it would lock up because it could not keep up.  I had to "turbo" overclock it to even get acceptable and semi-stable performance.  And it is not even just bCNC, Universal G-Code sender also pegs the speed governor at 100% and that is with no graphics!  I am starting to think that a RaspberryPi 2 is the minimum.  That is if you want to do any real work.  Some of the g-code files I was using today were over 200K and were running 1524 mm/m cutting speeds...

With that said, once I am happy with my configuration, I will post up the install script.  It is a Raspian install.  Has all the requirements for bCNC.  I also added noVNC, Samba windows file sharing and Inkscape.  Again, the RPi is WAY TOO SLOW to run Inkscape, but maybe the Pi2 will not.  I will let you know when mine comes in.

Bruce

Again, keep up the good work.  Probably not the place to ask this, but is your day job playing with a super collider?

vlachoudis, 2015-05-01:
You are right, I have to correct the gantry&probe for the proper units as well. I believe the problem is because you are changing the units inside the gcode (which is perfectly normal), which in this case I don't know what grbl will report back. The units as it is configured or the selected from the gcode. I will go to my garage to check with my CNC and do the correction.

Instead of the G92 use the G54 to 59 from the "WCS" tab. You are not forced to "zero"
the coordinates. You could set the real x,y,z position of the gantry e.g. z as 3.07mm on the settings below,  or in th TLO (tooloffset) settings. It would be persistent this way
You can even do it from the command line by typing setz 3.07

When you home your machine what WPos coordinates grbl gives you?

I understand the RPi is compact, but I really don't understand the benefit of it. I have a 10y old laptop as a controller from a garage sale (cheaper than the RPi!), it has a screen 4/3 (better than 16/9), wireless, usb, parallel, serial port, keyboard, power supply and with Ubuntu is way faster than the RPi. And I have almost everything installed in the 20Gb disk, inkscape, librecad, etc...

Indeed, my day job is simulating beam-matter interactions for the LHC and its injection chain, as well the development of the software physics simulation tools for it.

vlachoudis, 2015-05-01:
I had a bug in reading correctly the units (metric/imperial) setting from the init file. Nomatter what the CNC class was always reading it as metric, while on the Tools tab it was showing the correct setting. I've corrected on the new svn version.

Also after the "Stop" command I am issuing a soft reset to clean the remaining buffer in grbl. I don't know if it is the best way, but at least it will protect against accidental movements.

1bigpig, 2015-05-02:
Ok, you can give me back my hammer, because you nailed it!

Couple of things.  I am now getting some funky actions when I click the "Fit to screen" button.  Also, the WorkArea is still giving me some really large rectangles (like it is still be multiplied by 25.4).  It also seems like the tool animation is no longer being drawn.  Not a problem for me but just wanted to point that out.

The work area of the machine (and what and why grbl reports crazy negative numbers is set in $130 and $131.  It is negative for reasons that don't really make sense but apparently it has always been done that way, so while you have an option to make it report 0.0 for x and y after a homing routine.  As for what my machine reports, it reports -15.748 (400mm) in both X and Y after homing.  My machine has $13=1 set to report inch coordinates.

While I am not sure the popularity of the RaspberryPi, I have taken to one for CNC via a very long and winded disertation...but the skinny is this.  I have been using home CNC machines since about 1995.  Started with a Roland desktop mill and Macintosh //ci computer and later moved into parallel port machines such as MaxNC10-2 (still have that machine and now it runs grbl), a few ShopBots and Larken Automation CNC routers, a converted Mill/Drill and the list goes on.

Along the way, I was always using older computers as controllers but found that while they were cheap, they were unreliable.  Power supplys, fans, hard drives, monitors ect. would succumb to dust, sawdust, metal chips, spray coolant ect..  I live in a very corrosive environment of cat hair and high humidity.  Everything rust or corrodes here.  I decided that I wanted something small, that did not require the footprint of the PC, monitor and keyboard tied to every machine.  The grbl project was perfect as it provided a USB interface instead of a parallel port.  While there have been other, maybe better USB interfaced controllers--looking at you Mach3, they still required a PC, monitor and keyboard.  So, I have been using Universal G-Code Sender tied to a PC laptop, but it still involved exposing my laptop to sawdust and humidy...

So, I am interested in the Raspberry Pi computer because of no moving parts, being able to use them headless and having a low power draw.  I think my next project with be a box just big enough to house a Raspberry Pi 2 and an arduino Uno.  It with have an ethernet and power port on one end and a DB25 connector on the other.  With file sharing and remote desktop from any web brower, I can draw up my parts on my Windows design computer, copy them via Samba to my Pi and have it reside on the SD card.  Use bCNC to stream the the files to the Arduino and my tablet as a pendant to setup each job.

This is why the RPi is attractive to me.  I realize that it will never be a one size fits all, and I am ok with that.  Sorta how I have yet had a straight compile grbl.  All of them have required one change or another to suit my machine, my setup or my way of thinking.

Lastly, I have not yet had a chance to try the stop button.  Today was a stain and varnish day.  Maybe tomorrow I will get a chance to cut some more parts and I will let you know how it goes.

Bruce

1bigpig, 2015-05-02:
I wanted to make this separate, that way if you want it removed, it was be done without effecting the flow of conversation.

I figured you have to work some large organization with lots of employees/co-workers and that you have to be pretty important.  I saw your other project was a blue tooth proximity password override.  That takes someone who is very smart, had a need and is maybe kinda lazy--or at least hates doing repetitive things.

Second, I think that the amount of work and polish that bCNC has, its code base came from another of your projects.  I mean, if you did all of this from scratch in less than 6 months, my hat is off to you!!!

So, I decided to see if my hunches were right and Googled your name and CERN came up.  I figured that is where a really smart person like you would be.  Guess that is the bad thing about Google.  It makes it easy to track people down...

Thanks again for all you have done and all your time you have invested!  I know that sometimes it feels like you put all this work into it and only one or two people say "Thanks" and no one is using your program but I know for fact that while only one or two says thanks, MANY MANY more use your program daily and the reason why you never hear from them is because your program WORKS for them!  You only hear from people when things don't work...

vlachoudis, 2015-05-02:
@1bigpig  Fit to screen is now restored :) I accidentally destroyed when trying to delay zooming events to get a faster response on RPi.

I will check for the tool animation later. (I just woke up)

Google is great and dangerous! I have not uploaded on github all my projects :)
Part of the bCNC code comes from another project of mine [http://www.fluka.org/flair]

I appreciate a lot your comments.

vlachoudis, 2015-05-04:
@1bigpig  For me the gantry animation works well on both metric and imperial units. When you select imperial you have to set the inches in bCNC in the $13 of grbl and to ask the G20 in the startup commands.

1bigpig, 2015-05-04:
@vlachoudis I think it is working on the latest build.  I get the black circle that follows where the machine it currently "cutting".

I think this thread should probably be closed, but I have a whole new slew of questions...so I am going to open another issue.

1bigpig, 2015-05-05:
@vlachoudis OK, I lied. I checked my preview window, this time in ISO1 and no toolbit (inverted cone) is being drawn.
[image]

Unless I am missing something.

Bruce

vlachoudis, 2015-05-05:
@1bigpig I don't draw any inverted cone. Just a "red" circle you can see closed to 0,0.

chamnit, 2015-05-05:
@vlachoudis : If Python/tkinter works like Matlab, can you draw a 'v'? It's pretty much standard to see the inverted cone in CNC GUIs.

vlachoudis, 2015-05-05:
@chamnit no problem, its fast to do an inverted cone or a v

1bigpig, 2015-05-05:
@vlachoudis I must stop drinking...I swear when I first downloaded bCNC, there was an inverted cone 'tool' that followed the cutting path as the machine moved.  I must be loosing my marbles...

Bruce

Effer, 2015-05-05:
@1bigpig  Something like this?

[image]

It's Axis the GUI for LinuxCNC (or EMC2), another good choice for home DIY cnc.
http://www.linuxcnc.org/
http://linuxcnc.org/docs/html/gui/axis.html

But you need a PC with parallel port or expansive hardware. I find grbl a cheap and strong alternative.

vlachoudis, 2015-05-05:
@1bigpig I've converted the simple circle to an inverted cone (with lines not a nice 3d drawing) in the ISO projections

vlachoudis, 2015-05-07:
@1bigpig I think is fixed now all the problems, metric, imperial and inverted cone. So I close the issue

closed 2015-05-07 by vlachoudis