vlachoudis/bCNC #20: Probe readings

Effer, 2015-05-02:
Hi, I'm encountering some problems probing a non flat surface. From console I can see corret feedback from grbl but in the Gui all points are sets to zero.
I've try to manually edit points in .probe file and everything works as expected. The gcode file is interpoleted. 
Looking at the code I can't find where the probe response from grbl is parsed, maybe some internationalization issue? I'm using win 7 English but Italian location. Python 2.7. 
Best regards

vlachoudis, 2015-05-02:
@Effer Apparently grbl 0.9i has changed the return string of the probing. It added a suffix with :0 or :1 as a return code of the probing. I've modified the code to accept both the old and the new format.
Re download the code and should work. Please let me know to close the issue.

Effer, 2015-05-02:
@vlachoudis 
Perfect! Now everything works as should be.
The suffix :0  :1 means probe foult or probe completed correctly.
Thanks for fast fixing. And by the way, this project look very promising, keep on this way.

vlachoudis, 2015-05-02:
For the moment I skip the information :0 :1 I have to think what I do in the future with it.
Cheers
V.

closed 2015-05-02 by vlachoudis