vlachoudis/bCNC #22: M2 vs M5/M30

1bigpig, 2015-05-04:
@vlachoudis Got another question/problem.  I went ahead and modified my post processor in Aspire to process G-code for grbl .9i.  All is good and well except that I deleted the footer from 
"M30" to 
"M5"
"M2"

(M30 is program end and rewind --I guess back when they used to use punch tape, it would rewind the machine)
(M5 is spindle stop)
(M2 is program end)

So, with the new M5, M2, bCNC will hang on M5 and never release control back to the MDI/user unless to press 'Reset'.  I also tested this behavior with Universal G-Code Sender and it too does the same thing.

Now, if I remove the M2 from my post processor, as soon as it executes the M5 command, bCNC pops up a dialog and says "program executed in XXXX seconds..."

So, I don't know if this is a bug/flaw in grbl's M2/M30 implementation or if there is really something that bCNC and Universal G-Code sender want back from grbl after executing the M2/M30 command and are not getting it.  I am not sure.

Bruce

chamnit, 2015-05-04:
It's a bug in Grbl. I need to add a new state that called something like "Pause" or just put Grbl into a "Hold" state during an M0/2/30. After an M2/30, Grbl will soft reset. I need to come up with some clearer handling for it. But for now, just issue a '~' resume to continue.

1bigpig, 2015-05-05:
@chamnit Instead of creating a new state, why not just respond with "ok" and then reboot Grbl?  Seems like an easier solution and would solve the problem.  It seems both bCNC and Universal G-Code sender are waiting for the 'Ok' but since the rebooting, it never comes.

Just an idea...

chamnit, 2015-05-05:
@1bigpig : Grbl needs to prevent GUIs from sending more gcode immediately after an M30 and flush any queued data in the serial buffer. The pause happens to force an acknowledgement. Perhaps all I need is a message stating [program end] and require the GUI To send a ~ or soft reset.

vlachoudis, 2015-05-07:
@chamnit So I don't touch anything here waiting for the grbl fix

closed 2015-05-07 by vlachoudis