vlachoudis/bCNC #23: MDI clarification

chamnit, 2015-05-05:
Here's a simple request. 

Add `MDI`, called Manual Data Input in CNC circles, in front of the command line at the bottom of the GUI window. This will help clarify its use and help users notice it.

vlachoudis, 2015-05-05:
Good point! 

However the command line is far more powerful than simply sending commands to grbl.
You can do anything from the command line, like:
opening closing, reseting, homing, running, moving the gcode, rotating, cutting, profile calculations and many other (of course perfectly undocumented:)

Maybe called "Command:" to be more general is more appropriate

vlachoudis, 2015-05-05:
@chamnit Done as "Command:" if it is confusing I could change it to MDI:

Effer, 2015-05-05:
What about Command/MDI? It makes both.

vlachoudis, 2015-05-07:
Command: MDI in the tool tip :)

closed 2015-05-07 by vlachoudis