vlachoudis/bCNC #25: bCNC on Windows. Revert back to absolute mode after jogging.

chamnit, 2015-05-06:
A couple of ShapeOko users have noticed a some bugs:

For getting bCNC to run on Windows, they had some problems with a `rexx` library requirement. Here's a solution a user came up with that involves modifying the .bat file.
```
All you need is to add ./lib to your PYTHONPATH. Make a bCNC.bat with this contents:
CODE: SELECT ALL
@echo off
set DIR=%~dp0
set PYTHONPATH=%DIR%lib;%PYTHONPATH%
start python "%DIR%bCNC.py"
and run that instead of bCNC.py. It works with Python 2.7.4.
```

Next is reverting the G90/91 mode back to what it was before jogging was implemented. Often I see GUIs just send two commands to Grbl when jogging. First the typical, `G91 X0.01` to move incrementally, followed immediately by a `G90` to switch it back to absolute mode.

1bigpig, 2015-05-06:
@chamnit I have noticed a few other grbl front end programs (Universal G-Code sender) have the issue with the G90/G91 after a move command. I had to add a G90 command to my Aspire post processor header to make sure my cut files were always in the correct mode.  That is one solution, but I agree, it really should add a G90 after a move.

Bruce

vlachoudis, 2015-05-07:
@chamnit I've added the small bat, but I one needs to change the "set DIR=" variable before running
Also I've added G91 after every command. However since I am leaving for 10 days on an duty travel I don't have the possibility to check it.
I've made it as as two lines or I can remove the middle \n? would grbl accept it?
G91G0x%s\nG90\n

chamnit, 2015-05-07:
@vlachoudis : Thank you. The jogging command should be made two lines, because Grbl's parser will report a line with a G91 and G90 with a modal group error. I'll check it as soon as I can, if someone else doesn't beat me to it.

vlachoudis, 2015-05-07:
@chamnit Ok I've test it quickly on my machine and it works Ok.

closed 2015-05-07 by vlachoudis