vlachoudis/bCNC #27: GCode Error on circular move (G2)

Scott216, 2015-05-08:
I get an error when bCNC gets to this line:
G2X3.0898Y3.1724Z0.1176I-0.0891J-0.0785

bCNC error is: Invalid gcode ID:33

This gcode works fine on Chilipeppr.  My gshield is running GRBL v0.9g

Here's a link to the whole gcode file: http://pastebin.com/t44yqYg8
gcode was generated by Fusion 360

chamnit, 2015-05-08:
Ran this program on a stock Grbl v0.9i with the Grbl streaming script. It didn't report this error and ran fine.

Is bCNC doing something to the g-code before sending it to Grbl that would cause this problem?

Scott216, 2015-05-08:
I plan on upgrading to v0.9i soon. Before I do I need to make a new limit switch input board that uses D12 for the z-axis.
What did you mean by "with the Grbl streaming script"?

chamnit, 2015-05-08:
This program worked on Grbl v0.9g as well. And if you want variable spindle, then you'll need to swap D12 and D11 pins. Or you can keep your pin config, if you comment out VARIABLE_SPINDLE in the config.h file before you compile and upload GRbl to your Arduino.

Grbl officially supports only one streaming program and it's in the /docs/script/ dir, called stream.py. It's very simple and only streams, but it's very fast and serves as a benchmark.

vlachoudis, 2015-05-08:
I've tried with bCNC + grbl 0.9i and it stops in the line T1 M6 with an error 36.
If I comment the T1 M6 line the programs runs correctly with bCNC.

FYI For the moment I don't show to the user the correspondence of the returned "ok" with the line being executed, so the last line when you saw the error was the last in the buffer but not the one processed. I will implement it quite soon.

Before sending to grbl, bCNC strips the spaces and does a rounding to the number of digits the user wants configurable "Tools -> CNC -> Decimal Digits" the default value is 4 decimal digits (for metric units is more than enough). However in your case you don't have more than 4 decimal digits, so all commands are send, as is to grbl (space stripped only). This can be seen in the terminal that the lines send have exactly the same precession as the input

Effer, 2015-05-08:
@chamnit  M6 is not a supported command of grbl. Other streaming program (UGS for example) simply ignore "error: Unsupported command" from grbl. 
It think @vlachoudis prefer to put the state machine of bCNC in a safe state, stopping sending more gcode lines.
Could be usefull an option to manage the way of bCNC acts in case of unsupported command.

vlachoudis, 2015-05-08:
@Effer I could possibly add an option to ignore some types of errors/warning from grbl. I am not sure if it is a good practice to ignore them. I remember that @chamnit in another thread was mentioning that he plans to correct it in grbl.

chamnit, 2015-05-08:
@Effer : Yes, I hope I would know M6 is not supported by Grbl, as I am the developer. I wrote in the Grbl Wiki that GUIs should stop and force a user to acknowledge a problem in the g-code program, because this can lead to g-code modal states not being set correctly. @vlachoudis is following this suggestion. 

M6 is somewhat tricky. Grbl doesn't support tool changes, nor can it and arguably nor should it. So if Grbl simply ignores M6, then what do you do with the tool change? Skip it? You can't if you have to change a tool in the middle of a job. It's possible that I could mandate that GUIs look for M6 tool change commands, either to ignore them in a stream and simply pause the job when they encounter them or does Grbl have to have that responsibility. What about automated tool changes? The GUI has to be able to perform these tasks and Grbl has to be unpaused to do that. There isn't a simple answer.

Effer, 2015-05-08:
Wow I'm speaking with grbl developer! Sorry @chamnit I feel really fool now.
I'm quite just a beginer in grbl world. What sound strange to me it's that all other streamer I've try (UGS, grblPanel, the simply python streaming) simply throw every line they have to the serial port.
bCNC is not working in this way and could be really strange if you have some already tested gcode file that work with fine till now but you want to have a try with another streamer with some really nice features.

Also what often happen is that you generate gcode files from cam software that still not have good postprocessor for grbl. An option will make happy all people who like the risk ;)

Scott216, 2015-05-08:
I took M6 out and it runs fine now.  I'll try and update my Fusion 360 post processing file to not use M6.

vlachoudis, 2015-05-09:
Perfect. While waiting for @chamnit to implement the handling of these events, I will try to report the correct line which cause the error.
I close the issue then

closed 2015-05-09 by vlachoudis