vlachoudis/bCNC #28: Move to 0,0,0

Effer, 2015-05-09:
Now the "move to origin" button call this line of gcode:
"G90G0X0Y0Z0\n"
Grbl move in a straight line towards the origin, often this can lead to collision with material worked.
It would be useful to move Z to a safe travel height, than move X and Y to 0,0 and only than move down Z to 0.
Z safe could be saved in user settings file together with [CNC] other data.

chamnit, 2015-05-09:
That depends on where you've set your machine zero and how you use your machine. bCNC's jog center button for "move to origin" should be thought of move to work position zero. I think this should stay how it is. For the most part.

Typically the "go-to-home" (not machine home) that you want uses one of the G28 or G30 commands (the other is usually for a set tool touch plate position for tool changes). This moves the machine to a user-set position that is independent of the work coordinates, like the center of the X-Y workspace and Z at the top. This command is often used to move to the center of the workspace immediately after homing and immediately after a job so the spindle is out of the way. With a G28/30 command, you can also have it do an initial move, like Z+ to get the spindle out of harms way. 

That said, the jog center button could have a user-configurable string so they can enter `G90G0X0Y0\nG90G0Z0\n` if they need to.

But there should probably be another button for one of the G28/G30 predefined positions. These also can be made to have user configurable strings so that they can move the Z-axis first before moving over to the position like so `G28 Z5` to move to Z=5 in machine coordinates before moving in a straight line to the G28 stored position. See [link](http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G28-G28_1).

Effer, 2015-05-09:
Custom gcode would allow a wide range of personalizations. This would fit everyone special needs.
Some user button would be a good addition too. One could choose a smal label for the button and some line of code to send.

vlachoudis, 2015-05-23:
@Effer I've added custom buttons user configurable.
Also the [O] button to go to zero can be customized.
To customize the buttons you right click on the button and enter any commands you want as you would type it in the Command/MDI line.
The number of custom buttons is configurable in the .ini right now is set to 6
[O] + 5 buttons below

vlachoudis, 2015-05-23:
@Effer BTW the icon refers to all icons located in the icons/ folder with .gif extension.

Effer, 2015-05-23:
Is christmas time again? That's great @vlachoudis, a really clean implementation for a powerful feature. Thanks!

closed 2015-05-23 by Effer

chamnit, 2015-05-23:
@vlachoudis : Awesome! I just accidentally came upon this new feature by right-clicking on the O button. It's so nice that I can have it use `G28`.

colinluthier, 2018-06-17:
This is really helpful. Is there a way to have ore than 6?

vlachoudis, 2018-06-18:
In the ${HOME}/.bCNC file in the section [Buttons] add a ```n=###``` the number of buttons you want to have