vlachoudis/bCNC #29: Don't show probe grid until necessary

Effer, 2015-05-09:
I believe it's not necessary to show the yellow probe grid until probe data is setted o a probe cycle is started.
It only add more lines to main canvas.

chamnit, 2015-05-09:
I agree to this. I don't use auto-probing in my work flow, as it's better to have a square machine. This is usually limited to PCB milling.

Effer, 2015-05-09:
I use probing, I think autoleveling is one of the most powerful feature @vlachoudis implements on bCNC. I use it to v-carve over non flat surface of raw wood piece.
But for who don't use it only steals space.

vlachoudis, 2015-05-09:
You can always disable the display of the probing lines from the Menu->View->Probe, the setting is persistent,  it is recorded in the ini file.
However once you are setting the probing you need to visually see at which points it will probe over your object.
I could potentially remove it and display it only when the "Draw" button is pressed.

Effer, 2015-05-10:
Now I understand what is the draw button. 
Why not remove that button, and Draw the grid if the user change some of the autolevel settings. Less buttons means less documentation ;)

Also could be usefull to avoid probing points where there is no gocde. I mean: a point surrounded by square of the grid without g command path inside could be consider flat and not probed.

[image]

In this picture (courtesy of ScorchWorks) black cross are not probed.

 This will speed up the preparation. Here a link to this good software.
http://www.scorchworks.com/Gcoderipper/gcoderipper.html

Also if i save a probe data with the menu File->Probe->Save (not Save As) and the probe data was never been saved, it's not clear where and how the file is saved.
In this case what you think about saving the probe data with gcode file name but with .probe extension and in the same directory where gcode file was loaded? So the data will  be reloaded togheter if you load the file.
If gcode file was not loaded ask the user with the SaveAs Dialog.

vlachoudis, 2015-05-12:
The "Draw" button was a fast and dirty solution of my first try of the probe. I agree it should be removed. It has some techinical difficulties to avoid too many drawing events, but I have the solution of that.
The probe filename to be equal with the gcode file is a good option and easy to implement. The skipping of corner-points with that don't intersect with the gcode I have to check home fast it can be done with python.

vlachoudis, 2015-09-13:
@Effer Maybe I can put an option to enable the probe display only when the "Probe" tab is clicked

Effer, 2015-09-14:
@vlachoudis, what about simply disable the probe at startup? It already self enabled when setting some of the probe data.
It's just a matter to show a clean canvas on startup, especially for newcomers.

vlachoudis, 2015-09-15:
@Effer I've removed the restoring of the probe drawing flag. So by default is off on startup. I want to enable it only if probe data are loaded

closed 2015-09-15 by vlachoudis

Effer, 2015-09-15:
Nice!
Thanks.