vlachoudis/bCNC #30: Changing the view projection is not centered neither fitted

Effer, 2015-05-09:
Ideally changing the view should keep the same margin, and so allow to see the same part from another point of view. This not work especially changing from top o side view to ISO one.

Also if you select a part and then fit to screen the part is fitted, but if you deselect all (hitting ESC) and then try to fit again, it doesn't work. You need to Zoom out and then you can fit to screen.

chamnit, 2015-05-09:
I would like to add that the view window should probably auto-fit when you change views. This would cut down on one step.

Effer, 2015-05-09:
That would be good enough. Probably should be added also after opening a new file.

vlachoudis, 2015-05-11:
I found some time and added the auto fit on loading. While when changing views I will keep the same zooming level however I have to force the center of the screen to point to the same location as before.

vlachoudis, 2015-05-13:
Now when the view is changed it centers to the exact same location (x,y,z) at the center of the screen. It works nicely between X-Y and the ISO*, while it fail when switching to X-Z or Y-Z plots

closed 2015-05-13 by vlachoudis