vlachoudis/bCNC #31: Windows port auto-detection.

chamnit, 2015-05-10:
@vlachoudis : ShapeOko users have been trying out bCNC and most are using older Windows-based laptops to run it. For the most part, everything works pretty instantaneous once they install Python 2.7 and pyserial.

But the main issue is that they are having a lot of trouble connecting or figuring out what to do to connect to Grbl. The port detection defaults seem to only work for *nix system, as it works on my Mac and RPi Linux dist without a problem. 

Here's what they are currently doing to fix the issue.

```The default port name is not very helpful on Windows. Ports are named COM1 etc, not /dev/ttyACM0 etc. There's a method to enumerate them, but it seems bCNC is not using it. Just type COM5 (in your case) in the box to the left of the green Open button.```

Effer, 2015-05-10:
I'm using Win7 and python 2.7
In the control tab there is a combo box named connection. If i hit the drop down button there is a list of available serial port. One you choose the correct on the setting is saved.
[image]

chamnit, 2015-05-10:
The user is on Windows Vista. Here is a link to the forum thread. (http://www.shapeoko.com/forum/viewtopic.php?f=6&t=6086)

Effer, 2015-05-10:
I've just verify with Win Xp with python 2.7 and again Win7 but with Python 3.x and both it works.
May be he simply didn't see the small down arrow?

Effer, 2015-05-10:
Probably @vlachoudis should show the first serial port returned from serial enumerator at very first startup of bCNC, and not using the one from default bCNC.ini file.

chamnit, 2015-05-10:
It's possible that the controller's USB identification isn't compatible
with bCNC. It's a 328p-based controller that runs Grbl, but doesn't
identify as an Arduino. The company uses their own driver to help identify
when they are communicating with one of their products.

1bigpig, 2015-05-11:
@chamnit Shouldn't they be using Carbide Motion with their machine?  And, if the driver is not compatible, can't they just uninstall the proprietary driver and download the Arduino driver to make bCNC work with their machine?

chamnit, 2015-05-11:
@1bigpig : Sure, but they can technically use any gcode sender. It's stock Grbl. Carbide3d uses their driver so that their machine controller CarbideMotion doesn't have to be a generic sender. They can tailor it to their machines. I can't recall what they do, but I think it's in the USB driver. Instead of Arduino, it comes up as a Carbide3d device in the device manager (Windows). Right now I have no way to test this. I'm reporting the issues as I see them on the ShapeOko forums.

vlachoudis, 2015-05-11:
Hi all, indeed having as predefined variable the /dev/ttyACM0 only works for linux. When I made it I didn't have in mind other OS. Now I've change it to an empty string, forcing the user the first time to select the proper device from the dropdown box or manually input it.

Identifying the serial devices from python is a bit tricky. For the moment I am using the default enumerator of serial devices as it comes with the pyserial module and a home made one. I was looking the python pyserial implementation serial.tools.list_ports.comports() and it only detects the existence of a few predefined names. So if a device is identified with a non-standard name pyserial is not going to detect it.

vlachoudis, 2015-05-13:
For the moment I close this bug:
- I removed the predefined port. Now the user has to set it the first time he uses bCNC
- Concerning the port detection of "special" ports, either I will switch to my implementation and add all special ports or stick (as now) to the standard python one where the auto-detection will fail

closed 2015-05-13 by vlachoudis

tealvince, 2017-12-27:
FWIW, with the default code I've never gotten any ports to be detect under XP, but this works:

    def comports():
    	locations=[	'/dev/ttyACM',
    			'/dev/ttyUSB',
    			'/dev/ttyS',
    			'com']
    
    	comports = []
    	for prefix in locations:
    		for i in range(32):
    			device = "%s%d"%(prefix,i)
    			# Doesn't detect ports in Windows XP
    			try:
    				os.stat(device)
    				comports.append((device,None,None))
    			except OSError:
    				pass
    
    	      # Detects windows XP serial ports
    			try:
    				s = serial.Serial(device)
    				s.close()
    				comports.append((device,None,None))
    			except (OSError, serial.SerialException):
    				pass
    					
    	return comports