vlachoudis/bCNC #34: Focus keeps going to G56 button

Scott216, 2015-05-12:
I've got bCNC v0.3 on my windows 7 laptop.  Sometimes the WCS buttons work fine, but right now I keep clicking on G55 and the focus just jumps back to G56.  This was right after I used the touch probe with this command:
G54 G20 G38.2 Z-2 F0.118
Then I jugged the z-axis up.  After the jog, the WCS button worked normally

vlachoudis, 2015-05-12:
bCNC checks the state $G and sets the active workspace. Normally when you click on the G55 it will set it and then it will read the value from the $G and focus it. Now why it changes back G56, most probably I do something wrong in the program. Right now I am not at home to check, but once I will be back I will give it a look to correct it.

Scott216, 2015-05-13:
When you do look at it, try it with the G54 G20 G38.2 Z-2 F0.118 command.
The problem is pretty  repeatable after using this command

vlachoudis, 2015-05-13:
I don't have access to my CNC for some days. Could you send me a copy&paste of the terminal messages so I could simulate the problem

Scott216, 2015-05-13:
terminal output is on pastebin: 
http://pastebin.com/RTJKXheW

vlachoudis, 2015-05-14:
Thank you I will give it a look

jarretluft, 2015-05-19:
@vlachoudis I noticed something similar when updating the pendant.  You may want to consider requesting a $G after sending the updated WCS command (or any other change that will affect the controller state, such as coolant on/off, inch/mm mode change, etc)
For Example: if the user wants to switch to G56 from G54, send G56\n$G\n to set the controller to use G56 workspace and then update your stored $G variable with the latest info, otherwise it will switch back to the old value (G54) still saved in your $G array next time it loops through to update the UI.

I did this in the updated pendant code, so if a user changes WCS in the pendant, it requests updated $G state info so that the next time a getState() is run, the $G array values coming back from the python app are as up-to-date as possible.

vlachoudis, 2015-05-21:
@jarretluft thanks. how can I pull it from your fork?

closed 2015-05-30 by vlachoudis