vlachoudis/bCNC #36: Feature request

1bigpig, 2015-05-13:
If bCNC is thinking about implementing a tool change "feature", I would like to add a "Merge" g-code file option to the "File" menu.  Yes, I have ulterior motives for this command, but it would allow two or more different files for different tools to be in bCNC at the same time.  Once a tool change command is encountered, then bCNC could go off and do whatever little pseudo tool change routine it implements. 

Thanks,
Bruce

chamnit, 2015-05-13:
@1bigpig : I think I spoke about this once. Tool changes can be tricky. You have ensure that the g-code parser state is preserved and restored whenever you inject commands that are not part of the CAM-generated program. G-code modes can get screwed up. I've been thinking about a "sandbox" for Grbl, so it will retain the last position and parser state. You then can command whatever you want in whatever modes. When you exit the sandbox, it'll restore everything. This idea may be applied to jogging, parking motions, and tool changes.

1bigpig, 2015-05-13:
@chamnit I personally don't really care about tool changing, but I like your idea of sandboxing the G-code machine state.  I think there are a few issue to address, but nothing that cannot be overcome.

What I want/hoping for is the ability to merge 2 or more G-code files together from different CAM softwares.  I know that you can easily do it in text or heck even a terminal prompt, but I want to be able to import multiple G-code "geometries" and move them around.

For example, I want to make a sign or a control panel for grbl.  I want to cut make the cut outs and button holes with say OpenCAD and then process that into G-code.  Now, I want to v-carve the text for said control panel.  If I use say f-engrave (www.scorchworks.com) to V-Carve the text, I can them  import that into bCNC and move it over top of the button cut-outs.

This may seem convoluted, but right now, no program lets you graphically manipulate geometry in g-code form that I am aware of other than bCNC.  So, this _should_ be a simple code addition for @vlachoudis .

What I would really love to see is @vlachoudis make a real CAM program based on bCNC.  I think that he has most of the work done, but just needs to add a DXF/SVG import and do some simple pocket, contour and maybe even some V-carving toolpathing and that would go a LONG way to covering 95% of what people are using their CNC machines for.

Leave all the drawing/CAD to Inkscape and OpenCAD and just focus on the CAM part.  I personally would also leave out 3D toolpathing as I think that would dilute the purpose.  I am thinking like the very early V-Carve Pro type program.  I think version 1 or 2 would only import geometry via DXF and EPS and would let you select which geometry you wanted to pocket, contour cut and V-carve.  

I know there is a fair bit of this type of software for Windows, but not much (it any) for Linux.  There are a lot of separate programs that will do one thing or another but nothing that combines all 3 into on program that does it graphically.

Bruce

chamnit, 2015-05-13:
@1bigpig : Ah. I admit, I didn't read the original post very carefully.  Interesting idea though. I can see the value of being able to visualize multiple jobs in one viewer and how bCNC's tool path manipulation can be used to its advantage. I do see that this could get complicated quickly and being a fairly specific workflow. There isn't a reason that this couldn't be separate program that generates a new combined g-code jobs for bCNC to run.

I do a lot of machining of metal and don't like it when external programs mess with the original CAM-generated program. Small round-off errors, if not properly handled, could load a smaller end mill improperly and break it. Something like this is particularly sensitive when doing high-speed machining tool paths. And, I generally have a knee-jerk reaction to programs that try to do it all, because they tend to not do any one thing very well.

1bigpig, 2015-05-13:
@chamnit While I agree with you on messing with already tested G-code programs, bCNC is currently already doing this.  Maybe not on a basic file, but if you adjust your file for a probed surface, then the original g-code went out the window.  All arcs are converted to short line segments and then the code is reinterpreted based on the new surface.

While I don't have a problem with this per se, I see that this is/can be lesser of two evils.  If you are using two separate programs to generate your g-code, who knows how consistent the two different files will be.

Second, I think this would be a "stop gap" measure until @vlachoudis or someone else comes along and creates said CAM program.

Now, I also agree with you to a point about the CAM software trying to do too much, but I do not think that simple path cutting (contour cutting--on, inside and outside of said path), pocketing and V-Carve cutting are too much for one program to do.  Almost all of those functions (except maybe V-carve cutting) to almost exactly alike and share a very similar code base.  By leaving out DRAWING and 3D SURFACING, you have made something that would be infinitely usable by the open software community and would be very similar to what V-Carve Pro (www.Vectric.com) was in the early days.  By making it open, @vlachoudis would not have to sole support people and as similar features (spiral boring, ramping, drilling cycles ect.) get requested, he would not have to be the only coder.  

Again, this is just a pipe dream for now.  I don't know if @vlachoudis is even interested in a project like this.  I decided that I am going to learn Python and see if I can start cobbling something together but I am not a programmer by trade, so I am not expecting miracles.  What I do know is that 99% of those features are already available in open software and with some serious cutting and pasting and lots of glue code, it can be done without starting from scratch.

Bruce

vlachoudis, 2015-05-13:
@1bigpig Just for you information, bCNC is already (since some time) capable of:
- importing DXF (from the load menu, open a dxf file instead of a gcode one). It automatically joins the paths and converts them to gcode. For the moment only the spline is not working. I typically used it in combination with librecad
- can do "profiling" of the imported dxf paths
- cutting
- drilling

I have on my TODO list to do the pocketing.
and I didn't have svg importing. It is doable, but I don't like it so much :) with the dependence of the dpi from the program

1bigpig, 2015-05-14:
@vlachoudis I am going to investigate those features now that I have a faster (relatively) Linux machine.  The old Raspberry PI could not keep up but I have been trying two different boards (Raspberry Pi 2 and the Imagination Creator Ci20) that can almost run Inkscape now--the old Pi could not!

I did find a program that claims to do V-Carving, written in Python but I was not have any luck with it last night.  So, I am going to play around some more today and see what I can do.

I will say that those CAM features are kinda hidden--guess I really need to read the complete bCNC wiki...

Bruce

chamnit, 2015-05-14:
@vlachoudis : I wonder if its possible to pull in some previous work by others in the community. @tbfleming has a great open-source 2.5D CAM project called [JSCut](https://github.com/tbfleming/jscut). It has a lot of great features like pocketing, v-carving, profiling inside or out of a line, and a 3d cut viewer. Although it's written as a web app, some of the core components are written in C++. 

Although it might be a lot of work to import any of these JSCut features, it can serve are as good reference. But, rather than re-invent the wheel, most GUIs just link to JSCut and use its API to import the generated gcode right back into their GUI program. bCNC could then just concentrate on the g-code manipulation part only, if needed.

tbfleming, 2015-05-14:
@chamnit : jscut uses C++ (compiled to js) for v-carving and for splitting paths at tab boundaries. Everything else is in js.

chamnit, 2015-05-14:
@tbfleming : Thanks for the clarification Todd! Just curious, is there anything restricting the use of DXFs in JSCut? (to remove the DPI dependency).

tbfleming, 2015-05-14:
@chamnit : Most DXFs I encounter coming from free tools have open paths instead of closed ones. I chose SVG since Inkscape and similar tools make it easy to create closed paths.

vlachoudis, 2015-05-14:
@chamnit @tbfleming To create closed paths from open segments in DXF is trivial. Initially I've tried to use inkscape, it has fantastic capabilities as a vector drawing program but is not so intuitive as a CAD editor. Creating paths or items with exact dimensions is rather tedious, while programs like qcad or librecad are simple and effective. FreeCAD could do also parametric drawings, but is a bit buggy. Inkscape object dimensions are a bit messy for accurate path operations, since they include the stroke size.

Anyway, I didn't had the intention of re-inventing the wheel but I didn't find any free CAM tool that could use DXF. I wanted to have a profiling operation and it was not that difficult to create the algorithm.

vlachoudis, 2015-05-21:
@1bigpig I've added an import function in the file menu (as well in the command line) to import after the selected path a new gcode or dxf file. If nothing is selected then it appends at the end.

closed 2015-05-28 by vlachoudis