vlachoudis/bCNC #37: Build number in about box

1bigpig, 2015-05-13:
@vlachoudis I know that you are fixing so many bugs and updating the code all the time, it might be time to add a build number or build date to the about box.  Or at least a text file in the bCNC folder with the build date and number.  I am having a hard time knowing if I have the latest build.  

What I am doing now is running a script that deletes my bCNC folder and then downloads the latest from Github.

Bruce
PS with all the changes, it might be time for a major revision number as well.  I am sure you are now at version .4!

vlachoudis, 2015-05-14:
@1bigpig My approach on the numbering scheme, is Version.Release
Release: increase when I have added new functionality
Version: when and ever I will do major restructuring of the code
I should introduce a minor-release number for bug fixes or better as you propose the build number. I have to figure out how I could insert it from github to the code.

But yes there were quite a few corrections that deserve a new release 0.4.

Effer, 2015-05-14:
A numbering scheme will be welcome. In my softwares I like to add it in the Title of the main windows to easly identify with users. Sometimes I also add an Hash number of important files (you can never know what users try to do before ask for help ;).

vlachoudis, 2015-05-14:
@Effer my problem is that I don't know how to extract the build, release or hash number automatically out of the git repository.

Effer, 2015-05-14:
@vlachoudis if I fully understand I've found this git commands:
git log -n 1 --pretty=format:"%H"
or
git rev-parse --short --verify

You could run a script or a call from bCNC to modify some file before commit. It's just an idea, python not been compiled doesn't help in this case.

vlachoudis, 2015-05-20:
@Effer I've tried the git log but it returns the hex commit number. It is not very intuitive to look at. It would be nicer to have some minor number changing e.g.  0.4.1, 0.4.2
I was reading that git doesn't have tags as in svn to include this information, so maybe you are right I could write a small script to modify the file before committing and increasing the last number as well adding the date.

vlachoudis, 2015-05-20:
@1bigpig following the proposal of @Effer I've made a small script that I run before committing and it updates the version number and the last change field in the about dialog

1bigpig, 2015-05-20:
@vlachoudis Awesome!  You are the best!

closed 2015-05-21 by vlachoudis