vlachoudis/bCNC #39: Modify file externaly

Effer, 2015-05-15:
I can remember that at a certain point modify the file externally popup a message in bCNC asking if you want to reload the file. I've thought it was a nice feature the only thing missing was fit2screen.
But with last versions is not working anymore and the main canvas goes void, you need to manually fit2screen. I'm testing offline not sure if it also happen with connection open.
Of sure should be avoided while streaming is running.

vlachoudis, 2015-05-21:
@Effer it is still there. I tried simply to make it a bit smarter :) If it sees that the file loaded file is untouched but the file on disk is modified then it loads it automatically, otherwise it will ask.
The canvas going void might point to a bug. I will check it.

vlachoudis, 2015-05-21:
@Effer Ok there was a bug resetting the filename, so it was failing to open anything
I will push it in a while with some other changes

closed 2015-05-21 by vlachoudis