vlachoudis/bCNC #42: How to config WorkArea

Scott216, 2015-05-16:
How does the WorkArea view work?  When I bring in a file it looks like this:
[image]
I've got GRBL setup so when it homes, it sets x y and z to zero, not negative numbers like the default.

I posted the gcode I used in the above screenshot here: http://pastebin.com/X1P66uwV

chamnit, 2015-05-16:
I think it's based on the machine coordinate system and G92 offsets. Once you set part zero within the machine volume, the g-code paths should be inside the yellow box.

Scott216, 2015-05-16:
I don't use G92 for anything.  In this example I used G55 WCS.  After selecting G55 there was no change to the work area.  My gcode paths are still outside the area.

Effer, 2015-05-16:
@Scott216  working area is displayed based on some settings in Configuration file the fields are named **feedmax_x**, **feedmax_y**  and  **feedmax_z**   
The problem is that for some reason the values are readen in the opposite direction. So to have a try you can add a minus in front of the actual values. It seems also that the minus sign must be immediately after the equal symbol or will not be read correctly.  
I suspect @vlachoudis will fix this in a couple of nano seconds.

Have a look at <https://github.com/vlachoudis/bCNC/wiki/Configuration> on how reach the file based on your OS.

Scott216, 2015-05-16:
I'll give that a try.

vlachoudis, 2015-05-20:
@Effer the numbers are used as negative to describe the workarea, because this is what I was getting out from the grbl. Looking at the (thread)[https://github.com/grbl/grbl/issues/678] this is the default behavior of grbl. However @Scott216 after zeroing the G55 workspace it should come out right?

Effer, 2015-05-20:
That have a sense. Now I remeber that by tradition the coordinate system is referenced from top right corner. And at beginning was also the choice for my cnc. I will write some clarification in the wiki with a picture.

vlachoudis, 2015-05-21:
@Scott216 could you please confirm that by zeroing the WCS G55-G59 the workarea goes to the right place. If not then there is a bug somewhere :) which I need to find out.

Scott216, 2015-05-22:
It will have to wait until next week.

Scott216, 2015-05-26:
I just did a little test with the work area.  I use G56 for my work coordinates and loaded in a gcode file.  The work area is way too big, about 60 x 60 inches.  In the CNC section of the ini file i have about 11 x 13 inches.  The work area is oriented properly.  I'm using v 0.4.5
[image]

vlachoudis, 2015-05-27:
@Scott216 I forgot the correction of the workarea that @effer made. It should be corrected now

Scott216, 2015-05-27:
WorkArea seems okay now.  It's the right size and position

vlachoudis, 2015-05-27:
Perfect :)

closed 2015-05-27 by vlachoudis