vlachoudis/bCNC #43: Clear error in status bar

Scott216, 2015-05-16:
Is there a way to clear the status bar?  It seems once I get an error, it stays there, even if the error isn't relevant anymore and I'm going other things.

1bigpig, 2015-05-16:
@Scott216 I just press the "Unlock" button and the error reporter goes back to "Idle".  IF it is a more serious error, "Reset" will do it.

Bruce

Scott216, 2015-05-16:
I think it would be better if the status bar cleared on it's own once the error doesn't apply

vlachoudis, 2015-05-21:
@Scott216 The status bar is refreshed periodically and reflects the grbl status by sending a ? about 5 times a second to grbl to report the status. However if grbl reports an error or alarm then bCNC blocks all subsequent submissions until a Reset or Unlock is pressed as @1bigpig mentioned.

vlachoudis, 2015-09-13:
Again on the new interface <left-mouse-button> click on the status bar clears the error message.

closed 2015-09-13 by vlachoudis