vlachoudis/bCNC #44: Visualizer path color doesn't reset when job starts

Scott216, 2015-05-16:
I clicked on Control > Check gcode then went on to do other stuff.  Then I clicked Run expecting my Shapeoko to start milling, but bCNC was doing the gcode check instead.  It took me a minute to figure out what was going on.  In the Control menu drop down, if Check gcode is enabled you should have a checkmark next to it.  After it checked my gcode all the paths on the viewer were green.  Next I clicked Check gcode again hoping this would turn it off (which it did), then I clicked Run and the shapeoko started, but all the paths in the viewer stayed green.  I think the color of the paths should reset when you click on run.

Effer, 2015-05-17:
I agree, path already executed should be reset at Run click.

vlachoudis, 2015-05-20:
I've uploaded the correction. I have no access to my machine, could you check if it works properly.

Scott216, 2015-05-21:
I just downloaded the latest version v0.4.1 and tested this.  The color did not reset.  I enabled check code and ran it.  Lines turned green.  Then I turned off check code and ran my code for real, but the line color didn't reset.

vlachoudis, 2015-05-21:
Very strange! I cannot understand, I've just tested it on my machine and it works Ok.

Effer, 2015-05-21:
The same for me. But if you change point of view (from top to iso for example) the remaining path is reset to black. Also it doesn't reset the path neither if you restart the gcode after the end.

vlachoudis, 2015-09-13:
This is corrected in the new interface

closed 2015-09-13 by vlachoudis