vlachoudis/bCNC #45: Mouse position display in inches

Scott216, 2015-05-16:
When I move the mouse around in the visualizer bCNC displays the mouse position at the bottom of the screen, but it's in mm.  I think if I'm using inches, the mouse position should be inches also.

vlachoudis, 2015-05-19:
@Scott216 the visualizer shows the units in what you choose in the settings. The default one is mm but you can change it through the tabs "Tools" -> CNC -> Units
Then restart the program. I believe the suggestion of @Effer  is good to place also the units on the status bar.

Scott216, 2015-05-19:
If I change the units to inches, should I also convert all the other values in the CNC section, or will bCNC do that for me?

vlachoudis, 2015-05-19:
That's a good one! I was not sure what it will do, but testing shows that if you change to "inches" and close the program, then next time you open it, everything will be converted to inches.
Which proves to be a problem the first time someone changes the units and he thinks
that he can enter the units in inches, the program will still think in mm.
I have to correct it.

The program internally  and in .ini it writes everything in mm (I am thinking in metric :),
but it converts and displays to what the user selects.

closed 2015-05-20 by vlachoudis