vlachoudis/bCNC #49: Add a separate scale for the z axis in the control interface.

onekk, 2015-05-18:
Thank you for your work, I've been using it with my new eShapeoko with GRBL, and i have it running on an A20 Olinuxino Micro running debian wheezy (the stock image from Olimex plus some improvement).
I've done much mess and broken some bits, as I hit the Z axis control having the step set for a 100 mm travel, so i wonder if it is possible to have a separate scale for the z jog having only the 10,5,1,0.5,0.1,0.05,0.01 so a user can set the Z travel more in sync with the effective travel of the Z axis.

Many Thanks 

Carlo

vlachoudis, 2015-05-21:
@onekk You have a point there, but I am afraid of adding another scale will make the interface more cluttered and confuse people. I am thinking of adding a configurable maximum limit for the z-axis movement. So if the setting of the step is higher than this limit to always move equal to the limit amount.

onekk, 2015-05-22:
I think there is room just in the middle of the Z axis button.
Any manual or instructions on how to use all the functions in the program, i was wondering how to use the others functions.
I'm using it on an Olimex Olinuxino-A20-Micro, if you didn't know this board it's a little ARM based board with a version of debian-wheezy on it and it's running fine, for now I've only used for testing the machine and to level the wasteboard, so i need to put the machine in predefined position and test the heght with a dial caliper, to obtain the runot and shimming the bed in that point (using pop can and aluminiun foil scraps).
Now I'm wondering if i could write a script to run a drill job to bore the wasteboard to insert some t-nut on it to fasten the workpieces and the vise, i have inspected your code and it seems there are some functions.

Many thanks for your great joa and Best Regards

carlo D.

onekk, 2015-06-06:
I have just a added a pull request to implement this behaviour

vlachoudis, 2015-06-08:
@onekk thanks for the code. I understand your point of having a separate scale for z. Before adding it there are several small issues to be solved with such an implementation.
1. Keyboard bindings -+/*. I use them a lot and it is a fast way of altering the step
2. should be the same in the web-pendant as well
3. would it confuse the users having two scales?
4. the interface is already cluttered, adding more (even if there is space) will just make it worse.
5. maybe could be done with a flag to enable/disable such functionality
Did you try the [g]antry button to jog in the X-Y plane? It is faster and more accurate than setting a huge step and jog with the keyboard.
In any case i will give it a try after the eval version

onekk, 2015-06-08:
@vlachoudis, i think about your point:

1) I didn't know well the keybindings, (my fault)
2) I didn't use the web pendant, maybe i will try to implement it (but prior i have to buy a wi-fi adapter for the board)
3 and 4) the flag to enable/disable the functionality will solve 3 and 4, for 4 if you see it's only a little combobox and z.-step is separate in the behaviuor.

For the gantry button i have used only one time and it's great, and maybe set a custom button to move in a predefined position will also solve my problem.

Thanks and regards.

Carlo D.

Effer, 2015-06-09:
@onekk I was thinking if also a divisor for Z step could provide a good user experience for you?
I mean a ZStep ratio to use for reduce the incremental step of Z axis respect XY gantry.
In case of ZRatio=1 you'll move the same amount all three axis. 
Using a ZRatio=10 will move Z 10 times smaller respect X or Y.
If it works you'll get the required functionality at the cost of only one more widget. (the same as the optional combobox).

P.S. you can test the web pendant also locally by pointing your browser to http:\\localhost:8080

onekk, 2015-06-09:
@Effer It's a personal taste, i think it is useful the way I've implemented because I'm facing the problem is to set the z height in two scenarios:

1) in case of huge movement of the XY axis and if you are not concentrated and hit the Z up/down button / keypress, if you  reach the max z limit, you have to reset, rehome and so on, if you hit the piece you have to change the tool, and it will cost, so the amplitude of z movement is not related to a fraction of the XY movement say a default of 0.1mm is safe you are forced to set it manually to make damages.

2) when zeroing on the piece, or better when searching a clearance with the old method of using a sheet of paper to find the 0,01 height over the workpiece, you go down in 1mm increment then 0,5 then 0,01 until the paper is blocked by the tool, you raise it by 0,01 or 0,05 and you are done,  hoping the plane is at level, for example when milling PCB. In this case the movement are very "fine" and you work in the middle of the workpiece.

Obviously i think at the the hobby machines where the max z travel are in the order of 100 - 200 mm so an increment of 10 or maybe 25 is the max i have in mind for the scale.

But it's a personal taste, i agree with @vlachoudis that a preference (or other manner) to activate this function will solve all the issues (and make me happy).

As soon as i have my CNC equipped with a decent spindle (I'm waiting for a mount for my new Kress 1050 FME-1) i will make more work, for now I have done some test engraving, and lot of measuring with the micrometer to level the spoil board. (and so many z movements)

For the web pendant, my CNC controller didn't have a decent web browser as i use it only to read local html file (No internet connection where the machine is)

Effer, 2015-06-09:
@onekk now I understand better your needs. 
My suggestion clearly doesn't satisfy your way of work. 
I find useful your description. Compare bCnc users methodology help me learn more of cnc knowledge. Not only code worth to share :)

vlachoudis, 2015-09-13:
The new interface has as an option a separate z-scale as you requested. You can enable it if you add a "zstep = 1.0" in the [Control] section on your .bCNC ini
By default is disabled.

closed 2015-09-13 by vlachoudis