vlachoudis/bCNC #50: How to use the TLO function with a tool probe?

onekk, 2015-05-18:
Thanks for your great work, the wiki didn't say anything, how to use the TLO with a probe like this http://cnc-plus.de/images/tool-length-setting-sensor.jpg

chamnit, 2015-05-18:
Right now you have to do it manually. Record the probe location. When there is a tool change, do another probe and record the new probe location. Then, send Grbl the tool offset between the two probes with a G43.1 command.

onekk, 2015-05-19:
Thanks,  I have found this post (it's your post) 

>>- Insert your first tool. 
>>- Move to a fixed probing location to determine the z-location of the tool tip. People often use G38 and G30 commands to move there, and usually use a touch-plate of some kind to touch off the tip of the tool.

What is the command to issue the probing, G38.1 ?
 
>>- Run the job until a tool change.
>>- Insert the next tool.
>>- Move back to the probing location (via G28 or G30) and determine the z-location of the tip of the new tool. 
>>- Calculate the z-position difference between the first tool and this tool.
>>- Send Grbl a G43.1 command with that difference.
>>- Run the next part of the job until the next tool change. 
>>- Rinse, repeat, and always calculate the z-position difference based on the first tool location.

Carlo

chamnit, 2015-05-19:
See [link](http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G38-probe) for probing g-code commands supported by Grbl.

closed 2015-05-23 by vlachoudis