vlachoudis/bCNC #51: Stop Run problem with streaming

Effer, 2015-05-21:
It had already happen to me time ago but I left because suspecting it was something wrong with my gcode.
But now I've found how to reproduce it.
If you run a program and then stop it, and then push run again after some time the streaming stops. The second time the program run I've noticed the orange buffer bar is smaller then the first time. So I suspect it's a problem of flushing the characters counter in the streaming routine.

vlachoudis, 2015-05-30:
@Effer I've tried it on my machine but I cannot reproduce it.

Effer, 2015-05-30:
I can reproduce in this way. 
Hit run, bCNC will fill the buffer of grbl, then hit stop **before** bCNC restart to stream data to grbl.
Now hit run again, (buffer orange bar will be smaller then the first time) and after a while the streamer will stop.
I'll make more test and try to  locate better the problem.

Effer, 2015-08-12:
HI @vlachoudis, do you belive this could be related to #84 ?
I've made some short test but I can't reproduce anymore with last version.

vlachoudis, 2015-08-13:
@Effer I doubt it, since the progress bar is set during the preprocessing before the run. I will close it for moment

closed 2015-08-13 by vlachoudis