vlachoudis/bCNC #54: When a file is changed during the operation bCNC raise a reload dialog that fails everytime.

onekk, 2015-05-23:
Scenario, I have loaded a Gcode file.
I open the Gcode file in an external editor to change it (for example for a recursive  substitution say to substitude  a z-0.05 with Z-0.07 for every Z-0.05) bCNC raise a dialog "do you want to relad the modified file" i select Yes and the it raise a dialog "Gcode changed do you wnat to save it" i answer no as i want to use the file on disk. 

As result the file is not loaded in memory and i have to load it with open file dialog.

Many Thanks and Best regards

Carlo D.

vlachoudis, 2015-05-23:
Thank you Carlo,

I've just corrected it, could you download the new version and try it out.
V.

onekk, 2015-05-23:
Thanks a lot, you are very fast, i love the configurable buttons, I was trying to implement them :-D.
Monday i will try the modified version on the real CNC, and let you know.

Now it lacks only of the Z configurable steps. (I was also trying to implement it but maybe you are more fast than me, touching someone else code is a very delicate operation)

Many thanks and best regards.

Carlo D.

closed 2015-05-24 by vlachoudis