vlachoudis/bCNC #55: Pause button requires one initial click.

chamnit, 2015-05-23:
I've noticed in both the pendant and the main GUI that the 'Pause' button requires one initial click before operating correctly. For example, you just startup bCNC and load and run a program. If you press 'Pause' once and for the first time, it doesn't do anything. The second time and after the job will pause and resume. 

It looks like the initial condition for the Pause button isn't set to send a feed hold, but rather a cycle start. Should the pause button only pause (send a feed hold) and not perform two tasks? Could the 'Run' button be use to resume the job instead?

vlachoudis, 2015-05-24:
You are perfectly right. Ever since the moment you mention it I though it was the web pendant making the problem, however I had a bug and I had inverted the states of pause/resume. Now it is corrected

vlachoudis, 2015-05-24:
BTW. I wanted to ask you why in GRBL you are using 2 pins for the pause/resume and not only one. I have added on my control box the two buttons but I found them confusing. One would have been easier and you free on pin.

chamnit, 2015-05-24:
It's simply they way machine controller are designed. Resume(aka cycle start) has multiple tasks. Primarily it's starting a program, but also resumes a feed hold and an M0/M1 program pause. The resume button is a bit incomplete in Grbl, as it only does the latter. There are likely other uses for resume that I'm not thinking of at the moment as well.

closed 2015-05-24 by chamnit