vlachoudis/bCNC #56: Terminal activity while idle

Scott216, 2015-05-26:
My Shapeoko is idle, but in the terminal tab every few seconds I see this:
[G0 G56 G17 G20 G90 G94 M0 M5 M9 T0 F0.1 S0.0]
ok

I'm using bCNC 0.4.5 and the latest version of grbl - just downloaded from github today.  I'm not sure which version.

Effer, 2015-05-27:
I belive it's an new feature added from @vlachoudis  to update the status from grbl see here:
https://github.com/vlachoudis/bCNC/issues/12#issuecomment-105593584

Scott216, 2015-05-27:
okay, that makes sense.

closed 2015-05-27 by Scott216

vlachoudis, 2015-05-27:
@Scott216 could you check the latest commit. I've modified it to send the $G only when there is an activity, but not running and not during idle. I don't have access for a few days to my CNC to check it.

Scott216, 2015-05-27:
I've installed ver 0.4.7.  Now it's not sending anything automatically, only after I do something. So it seems to be working the way you want

chamnit, 2015-05-27:
@vlachoudis : FYI, in the next major release, I'm planning on having Grbl push the g-code state updates to GUI when they change, so you won't have to manage them. It'll be several months before this goes to master. Just wanted to let you know.

vlachoudis, 2015-05-27:
Thank you @Scott216 for the testing.

 @chamnit this would be perfect! It makes my life easier, since I wanted to find a way to send a $G only when there where commands that might affect the state.