vlachoudis/bCNC #58: How to get startup mode in inches (G20)

Scott216, 2015-05-27:
I don't know if this is a grbl issue or bCNC  When i first start things up, the units mode is metric (M21), how do I get default to be inches (M20)?

In GRBL I have 
$13=1 (report inches, bool)
$N0=G90 G20 G54 G17 

In bCNC.ini CNC section units = 0, but in the GUI, units is inches:
[image]

Here's my terminal window after startup.  I unlocked ($X), then Reset.  
Grbl 0.9i ['$' for help]
['$H'|'$X' to unlock]
$x
[Caution: Unlocked]
ok
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0.0 S0.0]
ok
Grbl 0.9i ['$' for help]
G90G20G54G17ok

chamnit, 2015-05-27:
Grbl defaults to G21 mm mode. This is the g-code standard default. To startup with G20 inches mode, Grbl has the $N startup-string commands to do this for you. To program it, send `$N0=G20` to Grbl. You'll see it execute every time you power up and soft-reset. (It DOESN'T do it when unlocking from an alarm. There are reasons for this. Just be aware.)

Effer, 2015-05-27:
@Scott216  are you looking the correct configuration file? In windows user file is located in %userprofile% and it's named ".bCNC". It's not the bCNC.ini file in program directory.

Scott216, 2015-05-27:
I was looking at the bCNC.ini file in program directory.

Scott216, 2015-05-27:
@chamnit I've got the $N0 set for G20 already, actually it's G90G20G54G17
I don't think it's running at startup, or if it is, something is changing it back to G21.
Here's what I did:
With bCNC running I closed my com port.  Next I unplugged the USB to gshield board for a few seconds.  After I plugged USB back in I opened the com port.  Then I tried to jog, but got bCNC error saying it was in Alarm lock.  So then I sent $x to the gshield.  Now I can jog, but it's in metric.  If I issue a home command it will go into inches.  From your post I understand $x will not run the $N0 commands.  When I plugged the USB cable back into my computer and the gshield powered up, shouldn't $N0 commands run and the gshield go into inches mode?

Scott216, 2015-05-27:
@chamnit BTW, I'm using grbl 0.9i, but I grabbed the github repo yesterday and uploaded using Arduino IDE.  So it's the version with the homing changes and some other stuff, M0 and M30 I think.

closed 2015-06-08 by vlachoudis

Scott216, 2015-06-21:
My shapeoko (gshield + GRBL v0.9i)  is switching from inches to mm after a reset.  Here's the scenario I just did.
Machine is in inches and I homed it.  I went to G0 X3 Y3, and it moved fine.  Then I did G0 X3 Y30.  This gave me a soft limit error as it should.  Then I clicked on Reset button then the Unlock button.  Next I did G0 X3 Y3, but this time it went to x=3 mm and y=3mm.  I understand from the above post that  $N0 isn't run on reset, but where is it getting changed from inches to metric?  My $N settings are:
$N0=G90G20G54G17
$N1=

from the .bCNC file in my user profile directory
[CNC]
units = 1
acceleration_x = 25.0
acceleration_y = 25.0
acceleration_z = 50.0
feedmax_x = 3000.0
feedmax_y = 3000.0
feedmax_z = 2000.0
travel_x = -287.0
travel_y = -325.0
travel_z = -70.0

Build date on bCNC is June 18, 2015