vlachoudis/bCNC #60: Macro Reserved Words for bCNC

1bigpig, 2015-05-28:
@vlachoudis I am proposing these as reserved works for a macro language.  The first is the RESERVE word, while the rest is the description and where that value is derived/found.  I tried to keep the words to 4 or 5 letters long.  I also tried to use the same language as the GRBL wiki.  So, if you see Active Mode, that is what the grbl wiki calls the response mode to a $G command.  I am sure there are a lot of things I missed, so I am open to suggestions and renames--if it makes sense.  Note, these would only return values in the Macro language.

Xcur			;Current X position in current work plane -Defined by ? command
Ycur			;Current Y position in current work plane -Defined by ? command
Zcur			;Current Z position in current work plane -Defined by ? command
Xmax			;Design/G-code Maximum X value -Calculated by bCNC
Ymax			;Design/G-code Maximum Y value -Calculated by bCNC
Zmax			;Design/G-code Maximum Z value -Calculated by bCNC
Xmin			;Design/G-code Minimum X value -Calculated by bCNC
Ymin			;Design/G-code Minimum Y value -Calculated by bCNC
Zmin			;Design/G-code Minimum Z value -Calculated by bCNC
Zsafe			;Z heigh that is predefined in Tools -> Stock -> Safe Z 
WC54X			;Work Coordinate G54 X value -Defined by $#
WC54Y			;Work Coordinate G54 Y value -Defined by $#
WC54Z			;Work Coordinate G54 Z value -Defined by $#
WC55X			;Work Coordinate G55 X value -Defined by $#
WC55Y			;Work Coordinate G55 Y value -Defined by $#
WC55Z			;Work Coordinate G55 Z value -Defined by $#
WC56X			;Work Coordinate G56 X value -Defined by $#
WC56Y			;Work Coordinate G56 Y value -Defined by $#
WC56Z			;Work Coordinate G56 Z value -Defined by $#
WC57X			;Work Coordinate G57 X value -Defined by $#
WC57Y			;Work Coordinate G57 Y value -Defined by $#
WC57Z			;Work Coordinate G57 Z value -Defined by $#
WC58X			;Work Coordinate G58 X value -Defined by $#
WC58Y			;Work Coordinate G58 Y value -Defined by $#
WC58Z			;Work Coordinate G58 Z value -Defined by $#
WC59X			;Work Coordinate G59 X value -Defined by $#
WC59Y			;Work Coordinate G59 Y value -Defined by $#
WC59Z			;Work Coordinate G59 Z value -Defined by $#
WC92X			;Temp Work Coordinate G92 X value -Defined by $#
WC92Y			;Temp Work Coordinate G92 Y value -Defined by $#
WC92Z			;Temp Work Coordinate G92 Z value -Defined by $#
ML28X			;Machine Location G28 X value -Defined by $#
ML28Y			;Machine Location G28 Y value -Defined by $#
ML28Z			;Machine Location G28 Z value -Defined by $#
ML30X			;Machine Location G30 X value -Defined by $#
ML30Y			;Machine Location G30 Y value -Defined by $#
ML30Z			;Machine Location G03 Z value -Defined by $#
TLOX			;Tool Length Offset X value -Defined by $#
TLOY			;Tool Length Offset Y value -Defined by $#
TLOZ			;Tool Length Offset Z value -Defined by $#
PRBX			;Probe X value -Defined by $#
PRBY			;Probe Y value -Defined by $#
PRBZ			;Probe Z value -Defined by $#
PRBS			;Probe Success state -Defined by $#
AMCG			;Active Mode Current/Modal G-code -Defined by $G
AMWP			;Active Mode Work Plane -Defined by $G
AMWC			;Active Mode Work Coodinates -Defined by $G
AMMI			;Active Mode Metric/Inch -Defined by $G
AMAR			;Active Mode Absolute/Relative -Defined by $G
AMFM			;Active Mode Feed Rate Mode -Defined by $G
AMPM			;Active Mode Program Mode -Defined by $G
AMSS			;Active Mode Spindle State -Defined by $G
AMTL			;Active Mode Tool Number -Defined by $G
AMFR			;Active Mode Feed Rate -Defined by $G
AMCS			;Active Mode Coolant State -Defined by $G

vlachoudis, 2015-05-31:
Thanks @1bigpig for the list. I have already started the implementation of the functions and it is helpful to see what I need. BTW are the AMxx variables somethink standard, otherwise I would prefer a bit more easier to remember names, like
"feed", "coolant", "plane",...
Also in my implementation since I am based on the python evaluation the variables will be case sensitive.

1bigpig, 2015-05-31:
@vlachoudis First off, it is your program, you can call them whatever you would like.  I am referencing the grbl wiki:
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#g---view-gcode-parser-state

Sonny calles the $G parser-state but the wiki uses the sentence:  "These active modes determine how the next G-code block or command will be interpreted by Grbl's G-code parser."

I would like to be careful and generic enough that I could do things like store the state of the G-code parser.  For example:

;Save the state of the G-code parser
< ;feed=[AMFR]>
< ;coolant=[AMCS]>
< ;plane=[AMWP]>
< ;unit=[AMMI]>
< ;absrel=[AMAR]>
< ;spindle=[AMSS]>

M5     ;turn off spindle
M9     ;turn off coolant
G91   ;switch to relative mode
G21    ;switch to metric

...do something cool like a tool change here...

;Restore the state of the G-code parser
G[feed]
G[plane]
G[unit]
G[absrel]
M[coolant]
M[spindle]

Effer, 2015-05-31:
There's an idea that I'd like to propose, that could be related to the coming scripting feature. it's about the possibilities to interact with the global flow of gcode.
It's consist to have bCNC **react** to some gcode **events**, for example when the program end or pause. This will allow bCNC to **run external script or executable**. So as example, it could be possible to automatically send a screenshot by email when work finish. Or may be an alert in case of errors. Or run a custom button code at specific line of gcode. Just an raw idea for now.

vlachoudis, 2015-09-13:
@Effer theoretically now with the function evaluation and python code execution you can execute any python command like
%import os; os.system("play ringtone.wav")
I didn't try it but it should work.
Otherwise I could introduce a
%system cmd
command to execute any system command

closed 2015-09-13 by vlachoudis

Effer, 2015-09-13:
Yes this works, but it is not sync with code execution.
It's executed as soon the line is buffered to grbl, this doesn't allow some useful application.

vlachoudis, 2015-09-13:
If you want to have it in sync with the code.
You should use the command "%wait"
The sender thread will wait until Grbl reports "Idle" as state

Effer, 2015-09-13:
Ok.
Just as reference:
%wait must be int the line before. Commands in the same line of %wait seems ignored.
Also in the editor if you add new line, the python code is executed. This should be avoided IMO.

vlachoudis, 2015-09-13:
Indeed wait should be on the line above.
Concerning the python code in the editor it is a tricky one! I need to execute them, because if they change the gcode, it has to plot it correctly. But how can it know if it is not indented to change the gcode? Maybe introduce another keyword to execute commands only during the running... hmmm...

vlachoudis, 2015-09-13:
If you see in the code CNC.py method parseLine2 (I have to change the name not very nice), there are a few special commands
%wait (without any argument, wait for the idle state of grbl)
%pause <msg> (display a message on status and put machine on hold)
%update <var> (update variable in the interface)
All other commands starting with % are send literally to python for execution