vlachoudis/bCNC #61: Feed under 1 not reported correcty

Effer, 2015-05-28:
If feed is under 1, State Feed label report  "0."
Also in case of round Feed values (as 100 for example) the label show an unsightly dot after the value (ex: 100. )

Greetings

vlachoudis, 2015-05-29:
@Effer, @chamnit I believe this is grbl issue. I am directly copying the string from the return of the $G command. Can you verify if the $G reports it exactly the same way as it is presented in the entry field.

Effer, 2015-05-29:
I'll verify later. (If I remember well terminal was correct, but I'm not sure).

Effer, 2015-05-29:
You was right @vlachoudis. I've made these two test from terminal:

G1 X100 F 0.1
ok
[G1 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok

G1 X100 F0.5
ok
$G
[G1 G54 G17 G21 G90 G94 M0 M5 M9 T0 F1. S0.]
ok

Something from status report of grbl is not as expected. I didn't verify if already documented in grbl site. Surely @chamnit knows better. ;)

vlachoudis, 2015-05-30:
@Effer Apparently it is a feature. For mm it reports with no decimal points and for inches it reports only one decimal point. The can be configured with
#define N_DECIMAL_RATEVALUE_INCH  1 // Rate or velocity value in in/min
#define N_DECIMAL_RATEVALUE_MM    0 // Rate or velocity value in mm/min
in the config.h. Maybe with the probing @chamnit has to increase the default values a bit, especially for the inches

closed 2015-05-30 by vlachoudis

Effer, 2015-05-30:
@vlachoudis Ok. Definitely not a bCNC issue.
Thanks.

chamnit, 2015-05-30:
Grbl has a lower speed limit. It has to do with floating point values approaching their precision limitsfor some unavoidable calculations. This limit is something like 0.1ipm.

vlachoudis, 2015-05-30:
Thanks @chamnit