vlachoudis/bCNC #62: Report: It works on an Olimex Olinuxino-A20-Micro board

onekk, 2015-05-29:
Only to report that bCNC is working great on a Olinuxino-A20-Micro board, It is a little board running a Slightly modified version of debian linux and the new debian Jessie support it natively, but i'm running the Olimex version based on Wheezy.

Thanks for your great job.

Regards

Carlo D.

vlachoudis, 2015-05-29:
Thank you @onekk . It would be nice to have a page on the wiki, with the OS that works and maybe special requirements. Do you want to help on that?

Effer, 2015-05-29:
Hi @onekk,
   feel free to add your experience here https://github.com/vlachoudis/bCNC/wiki/OS .
For now it's just a draft, any help is welcome.

Filippo

onekk, 2015-05-29:
Just added, let me know if it is clear and not too verbose.

Effer, 2015-05-29:
Well, @onekk definitely more detailed than what I was expecting. You run with default -v option ;)
Maybe last considerations are a bit too personal to be added in that kind of help page. But that's my opinion.

onekk, 2015-05-30:
@Effer ;) I know that here we are talking about a program written in a scripting language, but IMHO the final use of this program is to drive a piece of hardware so "It works" is not a useful report. In my "shop" i didn't have an internet connection so every missing piece of software, make me to disconnect the control box, load it on my shoulders and climb some stairs so it is tyring. :-D

vlachoudis, 2015-05-30:
Thank you @onekk.  I also like the documentation to be short and concise, however the additional info is fine. I believe It has to be organized in sections. First what it is needed to install and run the program, and any special issue, performance etc. Followed by the information for those who are interested on the auxiliary hardware/software aspects.

vlachoudis, 2015-05-30:
I forgot, maybe we should keep the documentation on the third person.

onekk, 2015-05-30:
Ok i will improve the page, thanks

onekk, 2015-05-30:
@vlachoudis have a look at the page and let me know if it's OK.

Effer, 2015-05-30:
Good work @onekk. It looks well to me. Detailed information are wellcome, it was simply a metter of style form. Thanks for sharing your experience.

onekk, 2015-05-30:
@Effer Sorry but I'm italian so I'm not very fluent in english writing.

vlachoudis, 2015-05-30:
Thanks @onekk.

closed 2015-05-30 by vlachoudis

Effer, 2015-05-30:
@onekk I'm Italian too ;D

onekk, 2015-05-31:
@Effer I've searched your site and seen where are you from, maybe in future when i pass near here (I own a vacation home quite near you) we can meet and drink a beer :-D