vlachoudis/bCNC #63: Feature request: Tool change

onekk, 2015-05-29:
It is possible to add a tool change functionality?

For example when in the Gcode file a Tx is found it is stored and maybe a message emitted, when a Tx is encountered but it is different from the preceding a message is emitted and the G-Code is stopped until the user hit a "done" button, maybe permitting to do some TLO and some homing (on some hobbyst machine the stepper motors are not energized and when you touch the spindle the machine will move from the position so a homing is necessary).
   
Regards

Carlo D.

vlachoudis, 2015-05-29:
@onekk it's on my todo list. I am currently implementing some basic scripting capabilities which would permit to make a macro for till change among others

onekk, 2015-05-30:
I'm waiting for it thanks.

kind regards

Carlo D.

vlachoudis, 2015-09-13:
The new interface now contains a manual tool change. Actually is user configurable to 3 actions, send the M6 as is to Grbl, skip the M6 lines or perform a Manual tool change

closed 2015-09-13 by vlachoudis