vlachoudis/bCNC #64: Feature request: it is possible to not pass to GRBL some Gcodes

onekk, 2015-05-29:
When using some gcode there are some code that are not managed by GRBL notably G61/G64 P...
It is possible to have a "blacklist" so if the Gcode file contains these code they were not pass to GRBL (ignored)

chamnit, 2015-05-29:
@onekk : This would a little difficult. You'd also have to make sure you include a list of supporting words that the command may require. It's best not to mess with g-code programs automatically, because it can cause more problems than solve. The proper place to do this is in the CAM post-processor. Not a streamer, IMO.

onekk, 2015-05-29:
I've seen in GRBL planner.c if I remember well a note about G61 / G64 telling that:
 NOTE: If the junction deviation value is finite, Grbl executes the motions in an exact path 
       mode (G61). If the junction deviation value is zero, Grbl will execute the motion in an exact
       stop mode (G61.1) manner.

so if i didn't go wrong GRBL is using G61 so why GRBL is not supporting G61? or I'm in error

Regards

Carlo D.

chamnit, 2015-05-29:
@onekk : It doesn't officially support G61.1 for reason that no one has explicitly requested it. As for G61 itself, I probably should add this as an inert g-code since Grbl v0.9 will give you an error.

onekk, 2015-05-29:
Many thanks
I'm using cammill (and trying to translate the interface in italian) and the postprocessor emit G61 or G64 according to a setting, I'm trying to write a also postprocessor for GRBL, so if it is safe to assume that GRBL is using  G61 i can simply omit the lines that print G61 on the Gcode and live with it. I guess that this is not affecting the generated gcode as it is a trajectory control on the corners (Tell me if I'm wrong assuming this)

Regards

Carlo D.

chamnit, 2015-05-29:
@onekk : I'll push an update with G61 as an inert g-code command later today. It's only a few lines of code to add this to prevent gcode parser errors, as G61 is the default mode for Grbl.

onekk, 2015-05-29:
Many many tahks

Carlo D.

vlachoudis, 2015-05-29:
@onekk you can tell bCNC to skip some commands by editing the ini file. For the moment in the default INI it skills the M6 tool change command. However as @chamnit said it's dangerous and better to correct from the cam side

onekk, 2015-05-30:
I think that the modify that @chamnit will solve the problem.

closed 2015-05-30 by onekk