vlachoudis/bCNC #66: Blue and Black toolpath colors

Scott216, 2015-06-04:
When I opened a gcode file, some of the toolpaths are black and some are blue.  What's the difference?
[image]

vlachoudis, 2015-06-05:
Blue are the selected ones. You can (un)select with the mouse clicking or dragging in the canvas or from the editor. Esc will unselect everything. Once you select you can process/modify the paths eg manually edit them, delete them, exclude them from sending, translate rotate, profile, and many other.

Effer, 2015-06-05:
On file opening nothing should be selected. But this is not working correctly. Also if you select some path and open another gcode file, the new part programm present some path already selected.
@vlachoudis  on an old pull request I've made a small fix to unselectAll on loadGcode in bCNC.py Maybe is gonna lost?

vlachoudis, 2015-06-05:
@Effer I don't remember this pull request. I will fix it now on the master stream.

vlachoudis, 2015-06-05:
Now it should be fixed

Effer, 2015-06-05:
It works better.Thanks.
Unfortunately still there are some defects on fit2screen after loading the file, especially if you open with path already selected. The gcode is not fitted, neither using the button. The only way to make it work again, is to zoom out a few time and then fit2screen.

vlachoudis, 2015-10-14:
@Scott216 @Effer is this issue still valid?

Effer, 2015-10-14:
Hi @vlachoudis, is still present the problem with fit2screen:
some time the calculation for adjust the screen view to fit the gcode give bad result. Mainly the zoom factor is correct but the offset for the scroll bar are not. I gave a look to the code but with poor results.

Effer, 2015-10-27:
@vlachoudis do you fix it? Fit2screen now seems to work. 
It was something related to the scroll bars of the canvas?

vlachoudis, 2015-10-27:
I corrected a few things but I didn't find the exact cause of it. It has to do something with the canvas widget of tkinter. 
I saw that sometimes the polylines are cropped if I don't draw the axes before. I believe that in some combinations the max limits of the canvas are not updated and then fit2screen doesn't work as well the zoom out.

Effer, 2015-10-27:
Anyway it works much better now.

closed 2016-01-16 by vlachoudis