vlachoudis/bCNC #67: The program is never terminated

onekk, 2015-06-05:
Maybe I'm doing something wrong, but when a program terminate and no g-code has to be executed  the progress bar is stalling and the interface is "blocked" unti I hit the "STOP" button two or three times (maybe the program is responding slowly as i use it on an ARM board)

chamnit, 2015-06-05:
Is it stuck at an M2 or M30 at the end of a program? If so update your Grbl to the most recent version.

onekk, 2015-06-05:
Ok i will try it will finish with M2 if I didn't go wrong.

Thanks

Carlo D.

chamnit, 2015-06-05:
@onekk : I meant that an M2 or M30 command at the end of a program can cause this perceived lock-up. Either remove these commands or update to the most recent build of Grbl to fix it.

onekk, 2015-06-05:
@chamnit, i have update grbl but the perceived lock is the information bar that remains green but now it show 100% and not 99% like in the prior version (I have just updated GRBL and bCNC on the machine).
A little note with build 29052015 (but i have synced my local copy of grbl today) the G61 is accepted and everything went fine. 

Thanks

Carlo D.

closed 2015-06-08 by vlachoudis