vlachoudis/bCNC #68: There is a way to find out what gcode line is causing an error in grbl?

onekk, 2015-06-05:
There is a proper way to track it? also a suggestion is useful so you can track down what "block" of code is generating the error.

TIA and regards

carlo D.

vlachoudis, 2015-06-05:
Not for the moment. It is on  my todo list (which is expanding all the time :) I have to implement some kind of logging in the sender routine and in case of error to report the line that is being executed.
Right now you can see on the terminal the last commands send to grbl, and the offending line will be there.

onekk, 2015-06-05:
Thanks, I didn't think about the terminal window. 
Your todo list is expanding as bCNC seem to be the most functional out here to drive GRBL, it is endorsed in some manner by @chamnit, and is a great program, maybe some refinement in the interface, in the everyday use, i would love to have:
1) a different scale for the z axis, it will be very useful to have say a 50 travel on XY to pull away the tool from the workpiece with the up/right button, and have say 5 on the Z to pull it out but not too far to have a little combobox will suffice between the two Z button.

2) a "DRO" more visible, more big character to see them from a more distance (for people like me who wear lenses it will be a huge improvement).

3) a reminder in the Control tab of which WCS is active (often homing became dangerous if you have issued a reset and you didn't remind that it have set the G54 WCS).

My two cents on improvements, i have bCNC running full screen 1024x768 (minus the mate panel up and down so i think quite 1024x700) on my Olimex A20-Olinuxino-Micor board with the VGA screen at 1,5 meter from the ground over my eShapeoko enclosure.

Many thanks for your kindness and for your great job.

regards

Carlo D.

chamnit, 2015-06-05:
@vlachoudis : I'm very grateful, and as probably everyone else is, for your work. I completely understand the ever expanding todo list problem as well. :) With open-source, I have to remind myself that it's my free time I'm are donating. So, don't feel obligated to do everything immediately.

vlachoudis, 2015-06-05:
You are welcome @chamnit. I don't feel obliged to do everything immediately, my todo list is sorted based on my priorities. Being a newcomer in CNC, the importance of some priorities can be changed from on the comment of some experienced users, like @1bigpig. My "handicap" is that if I see a nice idea (centering ring) I want to do it immediately otherwise in a few days I will forget :)

I've made my CNC to help me make things, not spend my free time for the CNC even though I must admit it is fun.

As you correctly pointed out the project is open source so everybody is more than welcome to contribute, as some users @Effer @jarretluft already did with code, documentation, design or ideas. BTW I want find a way to acknowledge their contribution.

Effer, 2015-06-05:
I'm comforted by the fact that the lack of free time is a common problem. 
@vlachoudis  I agree with @chamnit even if is fun, take it easy, or soon we will lose our BDFL. And that would be a really big issue ;)
It's a pleasure collaborate with you.

onekk, 2015-06-06:
@vlachoudis I have posted a merge request as a  "solution" for the #1 in my suggestions.
Regards
Carlo D.

vlachoudis, 2015-06-08:
BDFL I didn't know that term :)

onekk, 2015-06-08:
@Effer me too :-D

vlachoudis, 2015-09-13:
In the new interface, the machine status is a button and clicking on it it reports the offending line and an explanation of the error

closed 2015-09-13 by vlachoudis