vlachoudis/bCNC #70: Feature request: more visible DRO

onekk, 2015-06-06:
Hi to all how about a more visible DRO (in bCNC work and machine coordinates).
I have a little solution but in tkinter i didn't know how to make the fonts bigger than 14 points:

I have modified the bCNC.py adding:
self.dro_font = ("Helvetica,16")

and modified the code that describe the working and machine coordinates label as follows:

self.xwork = Label(frame, font = self.dro_font, foreground="DarkBlue", background="White",anchor=E)
		self.xwork.grid(row=row,column=col,padx=1,sticky=EW)

but i can't make them bigger than 14 points:

vlachoudis, 2015-06-08:
Thank you @onekk. I will add the fonts (user configurable from the ini) file once I complete the evaluation version. I am bit puzzled you cannot make them bigger than 14pt. If I remember well from tk8.5 it uses true type fonts, so there should be no problem with the scaling. Try as well negative sizes e.g. -16. Negative uses the true pixels of the screen and not converted from the DPI.

onekk, 2015-06-08:
@vlachoudis My main problem when developing is that bCNC isn't connected to the arduino it didn't show the "dro" value, and here when I have internet i don't have GRBL (is on the machine) maybe it will be useful to modify the code to make it possible to do a "test run" without having to be connected.

The code is almost done, but i i make a pull request it will be merged with that from the z-axis.
I'm waiting for the eval version with the scripting language, in the time i will try some modifications on the "real" installation of bCNC, i will try the negative values.

I'm studying your code as i didn't know well the tkinter interface, I've used wxwindows and pyside for my work with python.

Many Thanks and Regards
Carlo D.

vlachoudis, 2015-08-13:
@onekk I've implemented your suggestion on custom DRO fonts. You can configure them in the [Font] section of the ini file. For the moment I have only the DRO fonts but as I am creating in parallel the new interface I will push more fonts there

onekk, 2015-08-13:
Many Thanks, for now I'm not using the CNC, as I'm struggling with an image to gcode program to make it compatible with GRBL and to make it do the proper roughing cut without wasting time cutting air.

Regards
Carlo D.

closed 2015-08-14 by vlachoudis