vlachoudis/bCNC #74: Function key assignments

WillAdams, 2015-06-20:
It would be great to have bCNC map function keys to specific keypresses / functions in Grbl.

In particular, the direct Grbl commands would be great:

~ (cycle start)
 ! (feed hold)
 ? (current status)
ctrl-x (reset Grbl)

(Are there others?)

I’d also like to have commands suited for use w/ a physical pendant (I’ve begun using a Wii remote as one using Touchmote)

 - Homing
 - jog up, down, left, right, back and forth
 - increase and decrease the jogging increment
 - begin sending currently loaded file

vlachoudis, 2015-06-22:
It is certainly possible. Any idea which keys to use? or directly the same as grbl accepts?
I could add also keyboard mapping on the user defined buttons.

WillAdams, 2015-06-22:
My suggestion would be to capture the F# function keys (My Mac has F1--F15, but my PC keyboard only has F1--20).

Have a .ini file which allows one to assign a keystroke to them (or multiples would be great --- my Wacom tablet allows that), then one could set up a default mapping, but people could change it at need.

This way it would be invariant of whether or no a text field had the cursor focus, which would be really nice.

Fn keys seem to be typically arranged in blocks of four, so my suggestions would be:

F1 ~
F2 !
F3 ?
F4 ctrl x

F5 H (for homing)
F6 previous workspace coordinate system
F7 next workspace coordinate system
F8 begin sending currently loaded file

F9 jog X-axis left
F10 jog Y-axis away
F11 jog Y-axis near
F12 jog X-axis right

F13 jog Z-axis up
F14 job Z-axis down
F15 increase jogging increment
F16 decrease jogging increment

Having the feature, and having it invariant of keyboard focus and having keyboard commands for the things which we don't have thus far would be the most important thing. I've been using a Wii Remote as a pendant using Touchmote w/ Carbide Motion Machine Control Software and it works pretty well, I just wish they had a keyboard control to initiate homing, and commands for increasing / decreasing the jog increment as opposed to the current static 5 options which are hard to map w/ limited buttons.

onekk, 2015-06-23:
In the @WillAdams proposal why to split the jogging control to the F keys not usable by the PC users, and for the jogging I think that the arrow keys plus the page up ad page down for the Z axis are present in almost all keyboard, so i think that :

F1 ~
F2 !
F3 ?
F4 ctrl x

F5 - F6 - F7 - F8 - F9 - F10 - F11 - F12 user-configurable

So when a user has a keypad the jogging is easily done with the keypad with homing assigned to the 5 key (or maybe to the 0 key) with page up page down for the Z  and + and - for the increments.

Best regards to all.

Carlo D.

WillAdams, 2015-06-24:
My big concern about using the arrow keys for jogging is that they won't function for jogging the machine if the text cursor is active in a text field ---- maybe work this into the interface design somehow?

Agree the first four and user-configurable for the balance would be fine.

vlachoudis, 2015-06-24:
@WillAdams I had initially a global mapping on the arrows, but I found it rather confusing, when I waned to edit the text in the command line, editor or any text field. 
Hitting the Escape from anywhere moves the focus to the canvas/visualizer and the arrows work for jogging.

But it is rather simple to have in the configuration a list of mappings of the keys like the user defined buttons so any combination like Shift / Ctrl / Alt and any key F# or normal keys could be mapped to an action. I will keep in mind for the new interface

onekk, 2015-06-25:
@WillAdams, good point, i have noted this behavior but I've forgotten it, @vlachoudis Interesting the "Escape" trick.
I have forgotten the modifiers so also PC users have more then 12 F keys, the problem remain to remind all the mappings, maybe a little printed list or some stickers on the keyboard will ease the job.
But I'm going OT.

Regards to all

carlo D.

vlachoudis, 2015-09-13:
In the new interface all function keys are user configurable, in the Tools->Shortcuts

closed 2015-09-13 by vlachoudis