vlachoudis/bCNC #82: Error when connecting to GRBL with Mega2560

Stevenx1986x, 2015-07-12:
Every time I attempt to connect to my arduino it gives me the following error.
[image]
Has anyone else had and worked through this issue?

Effer, 2015-07-12:
Hi @Stevenx1986x 
Which OS are you using. And which python version?

Stevenx1986x, 2015-07-13:
I am running bCNC on Windows 7 x64 with Python 3.4.3, pyserial 2.7 for 3.4 and the latest master of bCNC. My microcontroller is also a standard Arduino Mega2560 flashed with GRBL 0.9i. All fusebits on the mega are untouched (in relation to startup timing bits), and I went through and erased the eeprom and reflashed Grbl last night just to make sure I could rule out any user errors.

Effer, 2015-07-13:
It could be that pySerial and Python 2 works different with Python 3.
I'll make some test later, there has been also some little adjustment on serial port lately.

Stevenx1986x, 2015-07-13:
If I happen to identify the issue i will be sure to report the solution back here. Very tempted to learn Python just to contribute to this project. Surely it cant be any more difficult than C++

vlachoudis, 2015-07-13:
Hi @Stevenx1986x, could you downgrade the bCNC.py. There was a recent change in the serial connection as @Effer mentioned to make it more robust, but maybe it doesn't work for python 3.
I will try to test it later on this afternoon at home.

Stevenx1986x, 2015-07-13:
I will try first thing after work tomorrow and report any changes.

vlachoudis, 2015-07-13:
I've changed the setDTR(True/False) to 1/0, since in the documentation it expects a level=<int>
maybe python 3.4.3 is more sensitive. Could you please try it and let us know if it works.

mandrav, 2015-07-13:
I 'm having the exact same problem if using the included bCNC script.
For the connection to work for me, I must edit it and change

"python ${DIR}/bCNC.py $*"
to
"python2 ${DIR}/bCNC.py $*"

(explicitely request for python2)

Effer, 2015-07-13:
It's definitely not working with Python 3.
Using self.serial.write("\r\n\r\n".encode('utf-8')) I can run through the open(self, device, baudrate) function,
but still the serialIO thread needs more fixes.
I've try some old versions but with same results. Seems like Python 3 is not as used as Python 2.

Stevenx1986x, 2015-07-13:
**@vlachoudis** issue is unchanged with the change. I will go ahead and downgrade bCNC and/or change the environment to use python 2 if that does not fix the issue.

Stevenx1986x, 2015-07-13:
Well I am not receiving the error anymore, but the MC is not responding to anything but a reset signal from bCNC. I am now using the latest version and have switched to Python 2.7
[image]

mandrav, 2015-07-14:
This is what I was getting every other time I was trying to connect.
The only way I could get that to work at that point was to disconnect bCNC, open arduino ide's serial terminal, verify the comms are working there (which they always were) and then re-open bCNC and comms would work again...
Are you sure you 're using the latest bCNC.py?

vlachoudis, 2015-07-14:
@Stevenx1986x apparently @Effer is right, there are several issues with python3. Strings in python3 are unicode while the serial interface requires byte-streams. I will try to convert all send commands to byte. @Effer it can be done using the b"\r\n\r\n" syntax which is accepted in python 2.7, instead of the .encode(), for all the hardcoded strings and it needs a bit of re-working the serialIO thread to properly convert the unicode strings.

Kawobei, 2015-07-15:
Well I wont raise the issue again.

Same thing but im Windows 10.

suprimex, 2015-08-11:
@Stevenx1986x 
absolutelly same issue. GRBL connected but not responding.  I see same as in your screenshot. Note that GRBL works for me in any other program but not with bCNC.

Mega2560
GRBL 0.9g

Lbuntu 12.04.5 LTS
Python 2.7.3

I do some checks with Python, looks like all is fine:
>>> import serial
>>> ser = serial.Serial('/dev/ttyACM0', 115200, timeout=1)
>>> s = ser.read(100)
>>> print s

Grbl 0.9g ['$' for help]
G21G90G17ok

>>> ser.write("?")
1
>>> s = ser.read(100)
>>> print s
\<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000\>

>>>

vlachoudis, 2015-08-13:
I have one mega2560. I will try to install grbl and test it

TilTisback, 2015-09-12:
Sorry to bump this again, but i have the same error.
Only i have an ATmega328P.

Effer, 2015-09-13:
Hi @TilTisback, which python version are you using?

TilTisback, 2015-09-13:
Python 3.4.3
I tested it now with the older version 2.7.10 and it seems to work fine.

Effer, 2015-09-13:
I think the code will be soon adapted also for python 3, unfortunatly it requires some adjustment inside some functions already under improvement in the eval branch.
So for now stuck on good old python 2.7

vlachoudis, 2015-09-15:
The new interface works nicely with ATMEGA2560. Python v3 is under way, so as @Effer mentioned we have to stick for the moment with good old python 2.7

closed 2015-09-15 by vlachoudis