vlachoudis/bCNC #83: TinyG Support

ymilord, 2015-07-15:
Would it be possible to add TinyG Support?

vlachoudis, 2015-07-17:
Maybe. I never used or looked TinyG. bCNC when was created was linked with grbl and focused to use the new features of 0.9 version. It would require to understand the differences of TinyG with grbl, commands communication protocol, etc... in order to see if it is possible to have support for both. For the time I don't have time to to work on that. If someone else is willing to do is more than welcome.

justinclift, 2015-09-09:
Yeah, this would be useful.

@vlachoudis TinyG is a fork of GRBL with support for 6-axis and other enhancements.  The people behind TinyG deprecated their own software sender solution a while ago, instead switching to rely on [Chilipepper](http://chilipeppr.com).  Since Chilipepper is an "online only" solution (eg you have to be online, signed in using a Google account, and so on, for it to work).  So, not a go-er for lots of people, and thus the search for alternatives :wink:

colinluthier, 2018-02-24:
This is still an issue. Is there any work being done to make bCNC compatible with TinyG?

vlachoudis, 2018-02-24:
Actually yes. I am doing some work on TinyG, I have a very preliminary version that the execution works, but I am missing some information, and I trying to find out on how to get it from the controller. Actually at the same time I am porting the whole code to python3. Most probably I will upload it in the next days in a separate branch for the brave ones to test it, and once it is stable I will merge with the master one.

colinluthier, 2018-02-24:
I would gladly beta test the code. I have an R7 that runs a few days a week.

justinclift, 2018-03-14:
@vlachoudis Co-incidentally, I've recently started to get back into CNC stuff.  Just flashed an Arduino Due board with TinyG2 (edge branch), and now looking for ways to control it.

Is the preliminary code for TinyG2 around somewhere?  I'm happy to try things out / report bugs / fix things that I can (etc).  Not doing anything critical here, am just adding a 4th axis to my old Shapeoko 3, thus the need for a new (non-GRBL) controller.

justinclift, 2018-05-07:
@vlachoudis As a data point, I've recently joined the g2core (official successor to TinyG) team [here on GitHub](https://github.com/synthetos/g2).

Apart from generally helping out there (eg cleaning up issues, expanding out docs) one of the things I want to get done is make g2core easier for people to get started with.  That means making it easier for projects like bCNC to add support for. :dancer: :tada: 

What's the info you're wanting to grab from the controller?  I can probably figure out how to get that done. :smile:

profitrolle, 2018-07-10:
It would be awesome to have bCNC working along with g2core :D (I don't say that because I hate js and love python...) Is there a way to get a look at or beta test the preliminary version?

justinclift, 2018-07-10:
@profitrolle Yep, this would be really interesting for a lot of people. :smile:

In the meantime, if you're looking for a G-Code sender for g2core then [CNCjs](https://github.com/cncjs/cncjs) seem to work well for people.

Universal G-Code Sender (UGS) is presently adding g2core support as well (it's now in their [nightly builds](https://github.com/winder/Universal-G-Code-Sender#downloads)). More info in [this UGS issue](https://github.com/winder/Universal-G-Code-Sender/issues/1071#issuecomment-403362488), if that's your kind of thing.

profitrolle, 2018-07-10:
Hi, I have followed you discussion when you where looking for an alternative. So you recommend CNCjs ? My other solution was to go for the Chilippepr Tablet workspace. I have no idea which one is the most stable and easy to tweak...

justinclift, 2018-07-10:
@profitrolle Yep, for something that works without much mucking around, I'd definitely go with CNCjs.

Not sure which OS you're using, but if it's windows then grab the appropriate 32 or 64 bit binary from their [Releases](https://github.com/cncjs/cncjs/releases) page.  I've used the 64-bit one there on Win7 and it worked well.

I also use CNCjs on my CentOS 7 x64 desktop (Linux), but I use the git version for that.

If you hit any problems, then open an issue on the CNCjs or g2core repo and someone is likely to help out in a reasonable time frame. :smile:

profitrolle, 2018-07-10:
Thanks for your response!
Well, i'm running some linux distro as well but the thing is running on a
rpi3 so it's raspbian.

As a complete beginner, is it possible to connect to the pi via my laptop's
webbrowser if the pi is running cncjs?

I find the documentation a little light for the moment but I'm still
digging...

(BTW you have convinced me, I'm in for cncjs... The installation is done)

justinclift, 2018-07-10:
> As a complete beginner, is it possible to connect to the pi via my laptop's webbrowser if the pi is running cncjs?

In theory yep.  I've not personally tried that (no rpi3 myself), but other people have and reported success. :smile:

Oh, with your g2core setup, is that using a Due board by itself for experimenting or have you got it hooked up to a CNC machine?

profitrolle, 2018-07-10:
Actually, i'm back on the project after a long break.

It is a rpi / due + g2core shield and the cnc /w controllers is connected
to the shield.

It was running like a charm but lacked some functionnalities so I'm getting
my hands dirty now.

And of course the setup is broken now... For unknown reasons I cannot
access my pi anymore (no ssh, nothing).

justinclift, 2018-07-10:
Ugh.  Busted rpi.  Hope you get that sorted. :smile:

With the Due if the g2core version is older than say 100.26, it's probably a good idea to update it.  eg to 100.26 or newer.  Saying that just to be on the safe side, as I know that's what CNCjs and others have been tested with.

scotthz, 2018-07-10:
@profitrolle Once you get your Pi up again, or especially if you end up reimaging your SD card, this is one way to get CNCjs running quickly without a lot of grief:

https://hub.docker.com/r/scotthz/cncjs/

My personal orientation is towards setups that are quick and easy to reproduce, hopefully mostly automated. I don't like to have to remember too much about everything I've done on a system, or to have to do it all over again to get a new one running. Using Docker helps in this regard.

profitrolle, 2018-07-11:
The rpi was fine :D it's my router that had an issue... 

The more I use cncJs, the more I like it! 

Have you already tried to make a custom widget? I have tried this afternoon but since I'm a complete noob in js/jsx/node/(...) I am kind of stuck. 
The only thing I have managed to do so far is to add a line to the existing customized widget to display some text...

justinclift, 2018-07-11:
Nope, haven't tried that at all.  I've actually been mostly focused on non-CNC related things (prioritisation :wink:), but I'll get around to it at some point.

It might be a good idea to open an issue about it on the CNCjs repo, as we're kind of using the bCNC repo here for non-bCNC stuff. :angel:

profitrolle, 2018-07-11:
> non-CNC related things (prioritisation 😉), but I'll get around to it at

Yes, sorry, i'll stop here.

Harvie, 2018-12-01:
I've separated controller support to individual plugins (#1020), so it should be relatively easy add another plugin for different motion controller.

justinclift, 2018-12-01:
Thanks @Harvie, that sounds really positive. :smile:

Harvie, 2018-12-03:
Problem is that i don't have tinyg nor the hardware to run it. Which also means that i don't really need this myself. So i guess that somebody will have to write this. Pull requests are welcome.

Everything needed should be available in "controllers" directory. I've just done some cleanup to make it easier to override some GRBL specific stuff. Look to the `SMOOTHIE.py` for example. it's based on `_GenericController` rather than `_GenericGRBL` class. However you will need to override some methods from `_GenericController.py` especialy the `parseLine()` method will have to be reimplemented, since TinyG uses JSON rather than GRBL/Smoothie style reports...

Harvie, 2018-12-03:
Does it mean, you can use TinyG using GRBL0, GRBL1 or SMOOTHIE driver in bCNC to run TinyG in text mode? (with exception of changing board settings)

colinluthier, 2018-12-03:
I don’t have an easy way to test that. My TinyG is in a production environment. I am switching it out in Jan for a GRBL device to get away from Chillipeppr. Once it is off that machine I can explore it more. The output to the TinyG should be able to remain unchanged. I am not sure about the status returned. I am pretty unfamiliar with the inner workings of all this. The TinyG has a big problem with G2 and G3 codes in inch mode and it’s best to not send them.

justinclift, 2018-12-04:
@colinluthier As a data point, Chilipeppr isn't the only GUI with good support for TinyG/g2core these days.  [CNCjs](https://github.com/cncjs/cncjs) should work well, as should [Universal G-Code Sender (UGS)](https://github.com/winder/Universal-G-Code-Sender). :smile:

colinluthier, 2018-12-04:
Thank you. I have not been able to get Universal-Gcode Sender to run on Linux. I have a Mac running it for a Arduino and a Linux box running chilipepr. Once the Tiny-G has been replaced on my main table I will likely use it for a smaller dedicated project. Perhaps bCNC will be ready by then.

bgbsww, 2022-12-14:
See https://github.com/vlachoudis/bCNC/pull/1772 for G2core support that might work with TinyG.  I can't test that case.