vlachoudis/bCNC #84: CNC takes off after clicking stop button

Scott216, 2015-07-19:
From time to time when I click the stop button in bCNC, my Shapeoko just takes off at high speed.  This is very bad.  It happened again today and I thought I'd try to reproduce the problem and document it.  Below is a YouTube link to the video showing this problem.  I also linked the gcode (on pastebin).  I'm running bCNC 0.4.9 June 18 and GRBL v0.9i.  Gcode was generated by Fusion360.

Video: https://www.youtube.com/watch?v=FgqczKtHxAo
G-code (generated by Fusion 360): http://pastebin.com/tr650GBC

Effer, 2015-07-23:
Hi @Scott216 some more informations about this problem? Did you find out how to reproduce it?

vlachoudis, 2015-07-23:
Hi @Scott216  I am far away from my machine to test it these weeks.  Just for confirmation you don't have any movement commands in the $N# intialization commands.

Scott216, 2015-07-23:
I did see the problem happen each time I clicked the stop button.  I only tried it a few times and I didn't try different gcode nor did I try rebooting.  
I don't have any move commands in the $N commands

Scott216, 2015-08-04:
This has happened several more times.  If I have to stop a job and the bit is down in a part, I disconnect the power to the motor driver board before clicking on the stop button to avoid the spindle taking off in x-y.

Effer, 2015-08-05:
@Scott216 I've recently encountered the same problem. Nothing strange in the terminal, no init code, tested with different gcode files at different feed. Can reproduce but not always.
I've try also try with older bCNC version and the same thing happen. What's strange is that I've never saw before. 
Only recents changes I remember are I've recently installed Python3 in parallel with Python2, then remove Python 3. I need to try with another host.

After the problem occur I can't send any more commands. Is the same for you?
I'm using windows 7 os, and you?

Effer, 2015-08-06:
I've made some more test:
Tried with a different PC and OS but same result.
It's not true that after the problem I can't send more commands, this happen only if I hit stop too many times. (another issue maybe)
My sensation is that the buffer queue to grbl is not completly empty and something remain when stopRun() is call. I suppose this becouse the end point is always part of the gcode.
I've try by adding some sleep inside the function but without improvments.
If you need more specific test let me know @vlachoudis .

Scott216, 2015-08-07:
When i have the problem, the machine is not making moves that match any of my gcode.

vlachoudis, 2015-08-10:
Hi @Scott216, could you download the latest version and try again. I've put an additional protection on the stopRun command

Effer, 2015-08-10:
Hi @vlachoudis , sorry but the problem persist.

vlachoudis, 2015-08-10:
@Effer I am not able to reproduce it on my machine. Could you send explain me the exact steps on how to repeat it.

Effer, 2015-08-10:
@vlachoudis here are my part programm: http://pastebin.com/wCjyX72g
I load the file and then hit the run button. I wait till about line 30 and then I hit stop. 
From the console this is a typical the output:
....
G3X0.5I0.5J0.375F360
ok
G1Y1.25
ok
G3X-0.5I-0.5J-0.375
ok
ok
G1Y-1.25
ok
G0Z6
G0X11.137Y-1.601Z6
ok
ok
Grbl 0.9i ['$' for help]
ok
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok

After that the machine stop and then after a second moves to another position in a straight line:

[image]

I can't send any user button command but I can restart again the gCode with run button.
If you're away from machine I doubt you can see the problem, from the console nothing strange to me.
Let me know if you need some more test.

Effer, 2015-08-10:
@vlachoudis in the serilaIO I've added some output to console in this way:

if tosend is not None and sum(cline) <= RX_BUFFER_SIZE-2:
	print tosend

and this:
def stopRun(self):
		print "FEED HOLD"
		self.feedHold()
		
		print "SLEEP 1"
		time.sleep(1)
		
		print "EMPTY QUEUE"
		self.emptyQueue()
		
		print "SOFT RESET"
		self.softReset()
		
		print "RUN END"
		self.runEnded()

in this way I've see that after the one second stop (after the feedHold command I presume) there is something send to grbl, and that is the line after.

Here another debug:

G1Y-1.25

G0Z6

G1Z-4.5F200

FEED HOLD
SLEEP 1
EMPTY QUEUE
SOFT RESET
RUN END
G3X0.5I0.5J0.375F360

Also the settings in my grbl if you need:

$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=4 (dir port invert mask:00000100)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=44.444 (x, step/mm)
$101=44.444 (y, step/mm)
$102=200.000 (z, step/mm)
$110=8000.000 (x max rate, mm/min)
$111=8000.000 (y max rate, mm/min)
$112=5000.000 (z max rate, mm/min)
$120=250.000 (x accel, mm/sec^2)
$121=250.000 (y accel, mm/sec^2)
$122=120.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Sorry if really verbose, I hope this help.

vlachoudis, 2015-08-10:
Thanks @Effer I think I have an idea on what is the problem.

vlachoudis, 2015-08-10:
@Effer I've made a correction, could you test it now

Effer, 2015-08-10:
@vlachoudis after some test it's seems working, but I'll have more tomorrow.
One thing I've notice is that after a couple of time I stop and run again the code, I canno't run the code anymore. bCnc seems like to be in undefined state and after hit Run nothing happen even if reset grbl, even with custom command.
For now thanks.

Scott216, 2015-08-10:
I tried the new code with your update from 30 minutes ago.  The machine hasn't taken off after hitting stop.  I'm running this gcode: http://pastebin.com/S19JTMDp
But I do have a problem after I click stop.  I can't jog or home.  Here's the steps
Click stop button
CNC stops, status = idle
click home - nothing happens
click jog - nothing happens
click run - CNC starts running
--------------------------------
Tried again:
click stop
CNC stops, status = idle
click unlock
click jog button - nothing happens
click home button - nothing happens
click Run - CNC starts running
-------------------------------------
Tried again
Click Stop
CNC stops, Status = idle
Click Reset
Click jog button - nothing happens
Click unlock
Click jog button - nothing happens
Click Run - CNC starts running

Scott216, 2015-08-10:
Here's another weird behavior:
1) Run - CNC starts running
2) Click Feed Hold button connected to Arduino 
3) Status = Hold
4) Click Stop button in bCNC
5) Status = Idle
6) Click jog buttons - nothing happens
7) Click Pause button in bCNC
8) CNC moves per the jog button presses in #6

Effer, 2015-08-11:
@vlachoudis  I've made some more test and the problem with erratic movement after the stop never happen.
As @Scott216 reported after hit the stop button the only way to work again is close bCNC and open it again.

vlachoudis, 2015-08-12:
@Scott216 @Effer I believe that the new commit has fixed all these issues. Please test it.

Effer, 2015-08-12:
Great @vlachoudis !
It seems ok to me. Start/Stop either with soft button and grbl inputs works as they should. And everything stop without strange behaviour.
Thanks.

Scott216, 2015-08-12:
Latest version is working well.  I don't have the problems listed above when I stop the job.

vlachoudis, 2015-08-13:
Great then I close the issue

closed 2015-08-13 by vlachoudis