vlachoudis/bCNC #85: Not quitting cleanly on OSX

AlexHolden, 2015-07-20:
Using bCNC on Mac OS X Lion, I can quit the program by using the 'X' window control to close the main GUI window, but not by selecting the Quit option from the menu bar or the dock icon menu. The keyboard shortcut command-Q doesn't work either. Instead of quitting, the program prints '0' to the terminal and the GUI freezes up.

vlachoudis, 2015-07-23:
Hi @AlexHolden, I don't have Mac to try it. What is puzzling me is that the [X] and the Quit call the same function. Maybe Mac is behaving differently than the PC. Could you provide some extra information (python version, tk etc.) you are using so maybe some other mac users can comment.

AlexHolden, 2015-07-24:
I just noticed there is also a File->Quit menu item; that one works correctly. I think the menu items that don't work properly (Python->Quit Python, [dock icon]->Quit, and the command-Q shortcut) are OSX specific and are being automatically generated by Python/Tkinter.

Googling around a bit, I found a question on Stackexchange from somebody else experiencing what sounds like the same problem, but his suggested fix didn't work for me (perhaps I was putting the line in the wrong place - I'm not familiar with Python or Tkinter).
http://stackoverflow.com/questions/5427875/python-how-can-i-make-my-tkinter-app-exit-properly

vlachoudis, 2015-08-10:
I've tried what he recommended, could you download the latest version and tell me if it works.

AlexHolden, 2015-08-14:
Latest git version still has the issue. It might be a problem with the native OSX port of Python/Tkinter; I get the impression that it's pretty old and not well-supported.
Don't worry about spending time looking for a fix on my account; I'm currently writing my own sender program using C/Gtk+.

vlachoudis, 2015-08-14:
Ok. @AlexHolden Just for future reference could you tell us which version of OSX python/tkinter you have

AlexHolden, 2015-08-14:
OSX 10.7.5
Python 2.7.1
Tk 8.5.9

closed 2015-08-16 by vlachoudis