vlachoudis/bCNC #86: wpos, mpos dont update

antoniosap, 2015-07-24:
this fields is empty, always

vlachoudis, 2015-07-24:
@antoniosap more information is needed to understand the problem. Which OS you are using? Grbl flair bCNC version? anything else that could provide some hint...

antoniosap, 2015-07-24:
bCNC 0.4.8 now is on raspberry Pi B+, gbrl is 0.9i. in graphic canvas the red tools dont move. the green path lines are ok.

antoniosap, 2015-07-26:
tested on bCNC 0.4.10. same

vlachoudis, 2015-08-03:
@antoniosap if the gantry (red tools) do not move, it means that you don't receive anything back from grbl. Please try to connect with a serial terminal to grbl and manually send ? and see grbl's reply.

antoniosap, 2015-08-03:
checked. grbl send back Mpos, Wpos and not fill the fields

vlachoudis, 2015-08-10:
Could you open the bCNC.py with an editor locate the "def serialIO(" method
and after the lines
`
					if line[0]=="<":
						pat = STATUSPAT.match(line)
`
add a statement
`
						print line,pat
`
You should see on the terminal a print out of the returned string from grbl and the pattern recognition object.

antoniosap, 2015-08-11:
<Hold,MPos:-36.750,9.040,25.000,WPos:-36.750,9.040,25.000,Lim:000> None
<Hold,MPos:-38.550,9.490,25.000,WPos:-38.550,9.490,25.000,Lim:000> None
<Hold,MPos:-39.730,9.780,25.000,WPos:-39.730,9.780,25.000,Lim:000> None
<Hold,MPos:-40.280,9.910,25.000,WPos:-40.280,9.910,25.000,Lim:000> None
<Run,MPos:-40.330,9.920,25.000,WPos:-40.330,9.920,25.000,Lim:000> None
<Run,MPos:-40.900,10.060,25.000,WPos:-40.900,10.060,25.000,Lim:000> None
<Run,MPos:-40.830,10.600,25.000,WPos:-40.830,10.600,25.000,Lim:000> None
<Run,MPos:-40.660,11.850,25.000,WPos:-40.660,11.850,25.000,Lim:000> None
<Run,MPos:-40.410,13.810,25.000,WPos:-40.410,13.810,25.000,Lim:000> None
<Run,MPos:-40.040,16.620,25.000,WPos:-40.040,16.620,25.000,Lim:000> None
<Run,MPos:-39.650,19.560,25.000,WPos:-39.650,19.560,25.000,Lim:000> None
<Run,MPos:-39.280,22.340,25.000,WPos:-39.280,22.340,25.000,Lim:000> None
<Run,MPos:-38.920,25.070,25.000,WPos:-38.920,25.070,25.000,Lim:000> None
<Hold,MPos:-38.650,27.150,25.000,WPos:-38.650,27.150,25.000,Lim:000> None
<Hold,MPos:-38.410,28.990,25.000,WPos:-38.410,28.990,25.000,Lim:000> None
<Hold,MPos:-38.300,29.780,25.000,WPos:-38.300,29.780,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None

antoniosap, 2015-08-11:
```
<Hold,MPos:-36.750,9.040,25.000,WPos:-36.750,9.040,25.000,Lim:000> None
<Hold,MPos:-38.550,9.490,25.000,WPos:-38.550,9.490,25.000,Lim:000> None
<Hold,MPos:-39.730,9.780,25.000,WPos:-39.730,9.780,25.000,Lim:000> None
<Hold,MPos:-40.280,9.910,25.000,WPos:-40.280,9.910,25.000,Lim:000> None
<Run,MPos:-40.330,9.920,25.000,WPos:-40.330,9.920,25.000,Lim:000> None
<Run,MPos:-40.900,10.060,25.000,WPos:-40.900,10.060,25.000,Lim:000> None
<Run,MPos:-40.830,10.600,25.000,WPos:-40.830,10.600,25.000,Lim:000> None
<Run,MPos:-40.660,11.850,25.000,WPos:-40.660,11.850,25.000,Lim:000> None
<Run,MPos:-40.410,13.810,25.000,WPos:-40.410,13.810,25.000,Lim:000> None
<Run,MPos:-40.040,16.620,25.000,WPos:-40.040,16.620,25.000,Lim:000> None
<Run,MPos:-39.650,19.560,25.000,WPos:-39.650,19.560,25.000,Lim:000> None
<Run,MPos:-39.280,22.340,25.000,WPos:-39.280,22.340,25.000,Lim:000> None
<Run,MPos:-38.920,25.070,25.000,WPos:-38.920,25.070,25.000,Lim:000> None
<Hold,MPos:-38.650,27.150,25.000,WPos:-38.650,27.150,25.000,Lim:000> None
<Hold,MPos:-38.410,28.990,25.000,WPos:-38.410,28.990,25.000,Lim:000> None
<Hold,MPos:-38.300,29.780,25.000,WPos:-38.300,29.780,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
<Hold,MPos:-38.300,29.800,25.000,WPos:-38.300,29.800,25.000,Lim:000> None
```

vlachoudis, 2015-08-11:
Thank you now it is corrected. bCNC was stopping at WPos: while you had enabled the Limit display
Now it should be fixed.

closed 2015-08-11 by antoniosap