vlachoudis/bCNC #87: Auto-Level Issue

Stevenx1986x, 2015-08-02:
I run my probe at the decently slow rate of 15 mm/min, but my machine allows starts to slope way too deep on one side to the point that it breaks my bits. I have broken more than 20 bits in the last 8 hours trying to figure out what the issue is, does anyone have a clue where I should start?

I clamp down, home the machine, probe the board (15x15 points per inch), go to about center of board, probe z, set zero, hit Z = 0, and then execute the g-code.

vlachoudis, 2015-08-03:
If I understand correctly, the problem appears during the g-code execution (with autoleveling) and not during the probing. There could be various causes for such behaviour: e.g., software problem in bCNC, probing problem due to bad contacts, electrical noise on the usb transmision cable, insufficient clamping (use double sided scotch tape for pcb), ...

 I want to verify if it is caused by the software, in this case I would require from you to send me your g-code, as well run a probing cycle, center and zero it, and save the probe data in a file (Do not execute the g-code). If you want send it to my email address (in the about dialog) and I will try to understand your problem.

akudoma, 2015-08-04:
This may be unrelated, but i too am having difficulties after auto-leveling. 

Version info:
Windows 7
Program     : bCNC
Version     : 0.4.10
Last Change : 18 Jun 2015
Platform    : win32
Python      : 2.7.10 (default, May 23 2015, 09:40:32) [MSC v.1500 32 bit (Intel)]
TkVersion   : 8.5
TclVersion  : 8.5

Grbl 0.9i
To avoid any broken bit's i am using a thick solid strand of copper wire. 

First i loaded the Gcode in for the board, zeroed the axis, and ran without auto-leveling. The full board runs perfectly. 

I run through the auto leveling steps, Get Margins, sett the min and max Z, set feed rate and the grid positioning, press draw to verify the grid, and scan. The scan runs well and the results are as expected. z deformations are in tenths of a mm. I chose a spot in the middle of the scanned area, and probe down along z. When the probe touches it stops. I press the set zero button for auto leveling. I also press set Z axis to zero.

When i try running the G code again, the spindle slowly takes off in across the board in a straight line and keeps on going forever, never stopping. I turned the steppers off and let it keep running to see if there was some kind of strange x/y axis offset that occurred but it didn't stop. It made it to (100,100) when the board is (60,40). 

I even saved the auto leveling, restarted the program, reloaded the gcode, verified the gcode starts to run correctly, loaded the auto leveling, and the same thing happens. The spindle just keeps traveling. 

I am very happy to send in my gcode and other other files you may want if you are interested in them.

akudoma, 2015-08-05:
I have been inspecting the raw Gcode in both instances and have noticed the program switches units and utilizes linear interpolation rather than rapid positioning. Here is the difference of the first few lines until the "lockup". The code may be shifted by one space, but i lined it up as best i could.

Working code | Z Adjusted Code
------------------|-----------------------
G20 | G20
G90|G90
G94|G94
F5|F5.00
G00Z0.01|G00 z0.0100
M03|M03
G04P1|G04 P1
G00X0.5425Y0.5732|G00X0.5425Y0.5732
G01Z-0.002| 
ok | 
G01X0.5444Y0.568|G1X13.7795Y14.55928Z0.173212
ok|ok
G01X0.5484Y0.5653|G1X13.82776Y14.4272Z0.172747
ok|ok
G01X0.5658Y0.5649|G1X13.92936Y14.35862Z0.172936
ok|ok
 |ok
 |ok
G01X0.5781Y0.5691|G1X14.37132Y14.34846Z0.175359
ok|ok
 |ok
 |ok
G01X0.5865Y0.5778|G1X14.68374Y14.45514Z0.177673
ok|ok
G01X0.5898Y0.5846|G1X14.8971Y14.67612Z0.18001
ok|ok
ok| 
G01X0.5929Y0.5967|G1X14.98092Y14.84884Z0.181363
ok|ok
G01X0.593Y0.6082 | G1X15.05966Y15.15618Z0.183371
ok| ok
G01X0.5898Y0.6211 | G1X15.0622Y15.44828Z0.184886
ok| ok
G01X0.5865Y0.6278 | G1X14.98092Y15.77594Z0.186152
ok|ok
G01X0.5774Y0.6369 | G1X14.8971Y15.94612Z0.186606
ok | ok
G01X0.566Y0.6407 | G1X14.66596Y16.17726Z0.18665
ok | ok
G01X0.5509Y0.6407 | G1X14.3764Y16.27378Z0.18572
ok | ok
G01X0.5469Y0.6397 | G1X13.99286Y16.27378Z0.183813
ok | ok
G01X0.5436Y0.6364 | G1X13.89126Y16.24838Z0.183168
ok | ok
G01X0.5425Y0.6323 | G1X13.80744Y16.16456Z0.182285
ok | ok
G01X0.5425Y0.5732 | G1X13.7795Y16.06042Z0.181566
ok | ok
|G1X13.7795Y14.55928Z0.173212
|ok
G00Z0.01 | G00Z0.01
ok|ok
G00X0.5594Y0.5817 | G00X0.5594Y0.5817
ok| ok
G01Z-0.002| G1X14.20876Y14.77518Z0.176761
ok| 
G01X0.5594Y0.6239| G1X14.20876Y15.84706Z0.182573
ok|ok
G01X0.5636Y0.6239| G1X14.31544Y15.84706Z0.183118
ok |Stops Hear, veers off forever.
G01X0.5689Y0.6222| 
ok| 
G01X0.5722Y0.6189| 
ok| 
ect...| 

I guess there could be several issues. 
(1)grbl not supporting G1 correctly
(2)Two commands are sent back to back without waiting for an OK and it seems to lock up after that
(3)Units seems to have changed but the header stayed the same, so wrong units?
(4)Some commands have different precision
(5)Some commands have spaces compared to the other G code

I will keep digging, but i thought i would pass this on. I am also hopping to dig into the code if i get the time and try to do some debugging.

akudoma, 2015-08-05:
Sorry for spamming the thread, but i have one more hint.
I edited the gcode to use G21(mm units) instead of G20 (inches) and although it was tiny, the mill ran through the whole board using the G1 commands just fine.

I think this is a unit conversion bug. The US really needs to switch to metric...

akudoma, 2015-08-05:
Confirmed, looks like a unit conversion bug. I just regenerated the original gcode in metric (units of mm) and the whole process went smoothly!

vlachoudis, 2015-08-05:
Thanks for the investigation! Indeed the interpolation was wrongly assuming that the probe reporting units are the same with the gcode. I will fix asap once I'll be back.

akudoma, 2015-08-06:
Happy to help! As a followup, i have been running gcode in mm units all day without a problem. Also, i have not yet had my spindle shoot off after hitting stop. I do lose connection with the board and have to reopen the port, but it hasn't moved after hitting stop for me.

vlachoudis, 2016-02-21:
With the new correction, it accepts mixed inch, mm gcodes

closed 2016-02-21 by vlachoudis