vlachoudis/bCNC #88: Undefined feedrate error

Scott216, 2015-08-07:
I'm getting an undefined feedrate error, but I can't figure out what the problem is.  I pasted my gcode on pastebin: http://pastebin.com/G6Ep09Ry
Can anyone help tell me where the problem is?

onekk, 2015-08-08:
I have had some problem like this.
I have found ou that if you have a G(0123) movement and then anothe type of G(0123) movement something happens and it raise the error, so I have resolved to put the feedrate on the movement change line. 
This code works, (I have used it with grbl 0.9i to cut some part).
Note that when i change the motion (the second line) I have to not specify the feedrate until i make a G0 move in the end and so on, but i prefer to prefix each movement with the appropriate G(0123) word, otherwise i get an error of undefined command (It strange but I'm struggling with it for almost a day prior to figure it out ) :

G1 Z-8.520 F250.000
G1 X1.277 F1500.000
G1 X1.489 Z-9.000
G1 X9.787
G1 X10.000 Z-6.402
G1 X10.213 Z-4.951
G1 X10.426 Z-4.402
G1 X12.128
G1 X12.340 Z-5.108
G1 X13.191
G1 X13.404 Z-4.873
G1 X14.255
G1 X14.468 Z-5.108
G1 X15.106
G1 X15.319 Z-5.304
G1 X19.149
G1 X19.362 Z-5.382
G1 X19.787
G1 X20.000 Z-5.304
G1 X25.957
G1 X26.170 Z-5.382
G1 X28.723
G1 X28.936 Z-5.265
G1 X31.915
G1 X32.128 Z-5.382
G1 X32.553
G1 X32.766 Z-5.461
G1 X33.404
G1 X33.617 Z-5.422
G1 X33.830
G1 X34.043 Z-5.382
G1 X37.447
G1 X37.660 Z-5.422
G1 X38.936
G1 X39.149 Z-5.304
G1 X40.000
G1 X40.213 Z-5.382
G1 X45.106
G1 X45.319 Z-5.304
G1 X47.234
G1 X47.447 Z-5.265
G1 X47.660 Z-4.245
G1 X51.064
G1 X51.277 Z-4.598
G1 X51.489
G1 X51.702 Z-4.873
G1 X51.915 Z-6.716
G1 X52.128 Z-9.000
G1 X57.447
G0 Z0.250 F5000.000

vlachoudis, 2015-08-10:
@Scott216 when I run your code I get the same error.
However the problem is a bit before. grbl reports
error: Invalid gcode ID:33
which means that you have an arc precision problem.
When an error is faced bCNC was calling the "Stop" method
to kill the current program. From a recent change this routine to stop a running program
was issuing a soft reset and then grbl was reporting "undefined feed rate"

I've correct that, and should be ok, now reporting only the Id:33 error

closed 2015-08-10 by vlachoudis

Scott216, 2015-08-10:
Thanks.  Can you tell me more about the arc precision error.  I'm not sure what the problem is there.

vlachoudis, 2015-08-10:
I am not in my garage now. I will check tomorrow to tell you the exact line. Most probable is that you have insufficient digits as precision or some incorrect rounding in an arc.

vlachoudis, 2015-08-10:
I've put a quick check in the code (I have not uploaded yet) and the line
G2 X1.9544 Y5.2935 I-0.0316 J0.0246
has an error of 0.5% of the (final end point - center) / radius
According to https://github.com/grbl/grbl/issues/619 it should be less than 0.1%
Maybe I will add it as an option to pre-check the g-code

Scott216, 2015-08-10:
How do you calculate the error?

vlachoudis, 2015-08-11:
radius=sqrt(i^2 + j^2)
error=abs(sqrt((x-xcenter)^2 + (y-ycenter)^2))-radius)/radius