vlachoudis/bCNC #92: Open file shortcut is enable during file processing.

Effer, 2015-08-15:
Using the shortcut Ctrl+O during the running of a part programm, pop up the open file dialog. I think this should be avoided.
Also in the canvas if you select a part (Ctrl+A) during the straming of the file you can you can move it.

vlachoudis, 2015-08-16:
I forgot to protect the keyboard shortcuts as well. I will put a protection in every function

vlachoudis, 2015-09-13:
@Effer I completly forgot that... :)

Effer, 2015-09-13:
@vlachoudis and so today we can declare the "close any issue day" ;)

vlachoudis, 2015-09-13:
Almost :) there are a couple ones still pending. I believe the most important is the Autolevel to check if the gcode program changes units during the run. But I wanted to do it carefully...

closed 2016-04-12 by vlachoudis